robotology/wb-toolbox #2: Migrating functionalities from old WBI-Toolbox

francesco-romano, 2015-11-30:
A lot of blocks must be re-implemented in this toolbox.

We should write a list and start implementing them

francesco-romano, 2015-12-04:
List of blocks to be migrated:

## State

- [ ] Joints configuration
- [ ] Joints velocity
- [ ] Joints acceleration
- [ ] Joints torques

## Actuators

- [ ] Position References
- [ ] Position (Direct) References
- [ ] Velocity References
- [ ] Torque References

## Model
- [ ] Joint limits
- [x] Forward Kinematics
- [ ] Jacobian
- [x] Mass Matrix
- [x] Bias forces
- [ ] `dot{J}dot{q}`
- [ ] Inverse Dynamics
- [ ] Centroidal Momentum

## Misc
- [] Smandruppator (@traversaro we need to do it parametric)
- [] Gazebo Synchronizer
- [] Real Time Synchroniser
- [x] Yarp Read
- [ ] Yarp Write
- [ ] MinimumJerk generator

cc @DanielePucci

francesco-romano, 2015-12-04:
Actuators done in 0be4acdce9ce7bb2f8defb7e0dcfaf6a60685c8c

francesco-romano, 2015-12-04:
Synchronizers done in e06151a7

traversaro, 2015-12-04:
The MinimumJerk generator is used by anyone? It would be nice to avoid depending on `icub-main`.

francesco-romano, 2015-12-04:
Yes. It is used to smooth the trajectories, e.g. in torque balancing

traversaro, 2015-12-04:
And I guess that making the dependency optional is not possible because the  `.mdl` that contains the block is unique, and cannot be "stripped" of some blocks on the cmake level, right?

francesco-romano, 2015-12-04:
Probably not at that level. Maybe we can think a bit about it...

francesco-romano, 2015-12-04:
yarp write added in ed78135dc5dcb61aef459e4b47ff45b5c9c87cbb

francesco-romano, 2015-12-04:
Jacobian added in 1871af9791dd7e424ccef9e5bdd6a858ff1e712b

francesco-romano, 2015-12-05:
Get estimate added in 5866eb327744a59436141dc5b54bc161b5a7b0b7

francesco-romano, 2015-12-05:
Inverse Dynamics added in  80d2dd5  

@DanielePucci and @traversaro I wrote a question in the commit message. When you have time can you express your opinion? Thanks :smile:

francesco-romano, 2015-12-05:
Added also joint limits and dotJ nu block

DanielePucci, 2015-12-06:
I have a question: if we parse the urdf to retrive the link list, we probably loose the CoM. Am I right?

francesco-romano, 2015-12-06:
I doubt the CoM frame is in the URDF. It is not a problem anyway to add it later.

Having said that, I am not a supporter of the list of frames. There are too many. You only get confused.

DanielePucci, 2015-12-06:
I agree. Concerning the `inverse dynamics block` can't we do a single block that takes as input `q, \nu, \dot{\nu}` and then it is up to the user to call it properly? Is seems to me that in this case we do not have to add any zeroing internally.  Am I right?

francesco-romano, 2015-12-07:
Yes, in this you are right. 
But bias forces are often used, and I would have liked to add a "shortcut" for the users, instead of letting them implementing it every time.

traversaro, 2015-12-07:
I agree that we need those blocks in the library. I would go for a simulink implementation (that is transparent to the simulink user, that can open and inspect the block) using the trick described in the commit message:

> One issue in this is how to define the zero constants for the joints (given that we do not know the size in Simulink but only in C++). We can multiply the Joint configuration input signal by a Zero-Gain. But this will of course affect computation (? maybe ?)

I have the impression that computations and communications (i.e. connecting blocks) in Simulink is rather efficient. There are no compatibility problems in having two instances of the same block in the same model?

francesco-romano, 2015-12-07:
Let's postpone the implementation of the robot mux/demux to a later stage.
As for the bias force, I'll switch to a pure simulink implementation with the "gain" trick.

closed 2015-12-07 by francesco-romano