robotology/wb-toolbox #5: Find a name for State

francesco-romano, 2015-12-04:
Currently the State component / block returns not only the state, but also other stuff (torques, etc).

You should find a proper name for state (and by the way also in the WholeBodyInterface)

cc @traversaro  @DanielePucci @iron76

DanielePucci, 2015-12-04:
I know you do not like sensors, guys

francesco-romano, 2015-12-05:
I do not like sensors, but maybe the others are ok with it.
Currently I called the block "Estimate"

DanielePucci, 2015-12-06:
`Estimate` :+1:

DanielePucci, 2015-12-06:
@traversaro ?

traversaro, 2015-12-07:
If I recall well, estimate was actually the initial name for the state interface. I don't remember why we then switched in some naming (the class name for example) to State. Estimates is definitely a better choice.

francesco-romano, 2015-12-07:
Great! Estimate decided then!

closed 2015-12-07 by francesco-romano