robotology/wb-toolbox #7: First  toolbox test

francesco-romano, 2015-12-07:
Before considering complete the first implementation we should test the toolbox.

I think the best way is to port the `torqueBalancing` controller from the old toolbox to this one.
@DanielePucci can you take a look at this on Wednesday?

DanielePucci, 2015-12-07:
Yes

DanielePucci, 2015-12-07:
Anyways, I'll start by migrating the `whole-body impedance`, then `torqueBalancing`

DanielePucci, 2015-12-10:
`torqueBalancing` migrated and tested. 
[image]

closed 2015-12-10 by francesco-romano

traversaro, 2015-12-10:
@DanielePucci  a short list of changes that are necessary to migrate the new WB version (transform all "model" blocks without inputs to take the input from the estimate block, ?? ) will be helpful for other people that needs to migrate.

francesco-romano, 2015-12-11:
@DanielePucci if you think it is worth the effort you can write a wiki page :smile:

DanielePucci, 2015-12-11:
Opening an issue on WBI-Toolbox controllers?

traversaro, 2015-12-11:
It would be better to have somebody that can be accessed by the entry-points (README/Doxygen) for WB/WBI-Toolbox documentation.