robotology/wb-toolbox #10: Announce WB-Toolbox to the CoDyCo community

traversaro, 2016-01-12:
Just an issue to track what we miss to have the WB-Toolbox: 
 * [ ] Simulink Library Integration (it is a blocking issue?)  https://github.com/robotology/WB-Toolbox/issues/1
 * [ ] Write a short "migration guide" to migrate a model from WBI-Toolbox to WB-Toolbox 
 * [ ] Finish porting the README from the old WB-Toolbox 
 * [ ] Announce the new release on GitHub with a notification to the group of all contributors and by email (additional channels could be a good idea). 

cc @jeljaik @francesco-romano @DanielePucci

francesco-romano, 2016-01-12:
-  About #1 : it is not blocking (in fact I removed it from the milestone 1.0)
- Maybe in the wiki?
- It should be ported. Isn't it?
- :+1:

traversaro, 2016-01-12:
Regarding the README : all the part on creating a new model using Gazebo is missing. Probably It would make to not having something of so advanced in the README, but we need to have it somewhere.

DanielePucci, 2016-01-12:
@gabrielenava is in charge of writing a `migration.md` file, to be included in the `/doc/` directory.

gabrielenava, 2016-01-13:
The first version of migration.md has been added in the /doc/ directory. Please tell me if there's something missing.

traversaro, 2016-01-13:
@gabrielenava I did a first cleanup of some links in https://github.com/robotology/WB-Toolbox/commit/8e32d34c628ca6a09695fe0655be024f87d5dc1f, some random comments : 
 * `rototraslation` is an italianism
 * I like the idea of a "Table of Content" that you added in the document, in that case is useful to add the link from the elements of the "Table of Content" to the sections in the document as done in https://raw.githubusercontent.com/robotology/idyntree/master/README.md . 
 * It may be useful to briefly describe the scheme that you put in the migration guide and what this scheme is doing
 * you forgot the `wholeBodyModel` in the TOC (table of contents).

traversaro, 2016-01-13:
Additional step before switching to `WB-Toolbox`: the controllers in `WBI-Toolbox-controllers` (especially torqueBalancing, that is mentioned in the codyco software main page) should be ported to WB-Toolbox.

traversaro, 2016-01-13:
Perhaps it would be a good idea to refer to it as WB-Toolbox 2.0 , so we state in two ways (new name and 2.0) that is the newer version of the WB**I**-Toolbox, because otherwise the difference given by just the **I** may be very confusing for most users.

francesco-romano, 2016-01-29:
I think that as soon the controllers are migrated we can announce the new toolbox.

traversaro, 2016-02-03:
@gabrielenava @DanielePucci what is the status of the controllers?

gabrielenava, 2016-02-04:
The simulink controllers have been updated with WB-Toolbox stuff since december. The joint space controller is still the version tested on the robot for my thesis, there are some things to do but not related with WB-Toolbox.

traversaro, 2016-02-04:
In https://github.com/robotology-playground/WBI-Toolbox-controllers/tree/master/controllers I see three controllers : 

 * `torqueBalancing`
 * `torqueBalancingOnSeesaw`
 * `wholeBodyImpedance`

Given that some of them have been updated the last time before the time `WB-Toolbox` 2.0 , I don't think they have been migrated. Can you double check?

francesco-romano, 2016-02-04:
We have to merge the other branch into master....

@DanielePucci

DanielePucci, 2016-02-04:
Yes, the migrated versions are under the branch  `torqueBalancingJointControl` to be merged as soon as the `YOGA` demo will be stable.

traversaro, 2016-02-08:
For Walkman guys we should also document how to compile the WB-Toolbox without `icub-main` blocks.

traversaro, 2016-02-14:
@DanielePucci Why we can't merge the `torqueBalancingJointControl` branch if the `YOGA++` demo is not "stable"? As far as I have understood it is already in the `torqueBalancingJointControl` branch, so it will be simply "unstable" in the master branch.

DanielePucci, 2016-02-15:
It is not only a matter of stability of the `YOGA++`. Configuration files for other robots must be added, and some clean up is needed. Do you really need the `YOGA++` in master?

traversaro, 2016-02-15:
No, we need to announce WB-Toolbox 2.0 to everyone (including people in our own lab by the way). 
This requires controllers (at least `torqueBalancing` and `wholeBodyImpedance`) to be ported to the WB-Toolbox. This has already happened in the `torqueBalancingJointControl`, and that is the reason why we need to merge that branch in master. 
The fact that `YOGA++` is ready was listed in the prerequisite for merging this branch, but I don't see why.

traversaro, 2016-02-15:
Ping @DanielePucci

DanielePucci, 2016-02-15:
pong @traversaro

DanielePucci, 2016-03-04:
Added configuration files in ffbf412fe27cda2bca69dde2e0e625a847d3eb76. Ready to merge.

@gabrielenava @traversaro @francesco-romano

traversaro, 2016-03-04:
I think we still need to fix issue https://github.com/robotology-playground/WBI-Toolbox-controllers/issues/17 .

DanielePucci, 2016-03-04:
Yes.  @francesco-romano

traversaro, 2016-03-18:
Final announcement done in https://github.com/robotology/codyco-superbuild/issues/114 .

closed 2016-03-18 by traversaro