robotology/wb-toolbox #11: Weird matlab crash

DanielePucci, 2016-01-21:
~~~
MATLAB crash file:/Users/DanielePucci/matlab_crash_dump.2284-1:

------------------------------------------------------------------------
              abort() detected at Thu Jan 21 14:29:45 2016
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled
  Crash Mode          : continue (default)
  Current Graphics Driver: Intel Inc. Intel(R) Iris(TM) Graphics 6100 Version 2.1 INTEL-10.12.13
  Current Visual      : Quartz
  Default Encoding    : US-ASCII
  Host Name           : Danieles-MacBook-Pro.local
  MATLAB Architecture : maci64
  MATLAB Root         : /Applications/MATLAB_R2015b.app
  MATLAB Version      : 8.6.0.267246 (R2015b)
  OpenGL              : hardware
  Operating System    : Darwin 15.2.0 Darwin Kernel Version 15.2.0: Fri Nov 13 19:56:56 PST 2015; root:xnu-3248.20.55~2/RELEASE_X86_64 x86_64
  Processor ID        : x86 Family 6 Model 61 Stepping 4, GenuineIntel
  Virtual Machine     : Java 1.7.0_75-b13 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : Quartz

Fault Count: 1

Abnormal termination:
abort()

Register State (from fault):
  RAX = 000000000000005f  RBX = 0000000000000000
  RCX = 0000000000000003  RDX = 00007000063515ac
  RSP = 0000000000000016  RBP = 00007fff85ec2d12
  RSI = 0000700006351590  RDI = 0000700006352000

   R8 = 0000000000000006   R9 = 0000000106eff000
  R10 = 0000000000000007  R11 = 0000700006352000
  R12 = 0000000000000000  R13 = 00007000063515c0
  R14 = 00007fff85ec35c5  R15 = 0000000000000000

  RIP = 000244db063515e0  RFL = 00007000063515e0

   CS = 00007fa2962eace0   FS = 00007000063515f0   GS = 00007fff924596e7

Stack Trace (from fault):
[  0] 0x000000010936aa34 /Applications/MATLAB_R2015b.app/bin/maci64/libmwfl.dylib+00027188 _ZN2fl4diag15stacktrace_base7captureERKNS0_14thread_contextEm+00000052
[  1] 0x000000010936e35a /Applications/MATLAB_R2015b.app/bin/maci64/libmwfl.dylib+00041818 _ZN2fl4test17terminate_handledEv+00000746
[  2] 0x000000010936de08 /Applications/MATLAB_R2015b.app/bin/maci64/libmwfl.dylib+00040456 _ZN2fl4diag13terminate_logEPKcPK17__darwin_ucontext+00000152
[  3] 0x000000010b0b7316 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00336662 _Z32mnRunPathDependentInitializationv+00002806
[  4] 0x000000010b0b75a9 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00337321 _Z32mnRunPathDependentInitializationv+00003465
[  5] 0x000000010b0b6c2a /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00334890 _Z32mnRunPathDependentInitializationv+00001034
[  6] 0x000000010b0b666e /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00333422 _Z22mnInstallCrashHandlersv+00000270
[  7] 0x00007fff8b3bdeaa           /usr/lib/system/libsystem_platform.dylib+00020138 _sigtramp+00000026
[  8] 0x0000000000000000                                   <unknown-module>+00000000
[  9] 0x00007fff924596e7                  /usr/lib/system/libsystem_c.dylib+00386791 abort+00000129
[ 10] 0x00007fff88a50070             /usr/lib/system/libsystem_malloc.dylib+00016496 free+00000425
[ 11] 0x000000014b901519 /Users/DanielePucci/src/install/lib/libYARP_OS.1.dylib+00521497 _ZN4yarp2os4impl17PortCoreInputUnitD2Ev+00000121
[ 12] 0x000000014b90127e /Users/DanielePucci/src/install/lib/libYARP_OS.1.dylib+00520830 _ZN4yarp2os4impl17PortCoreInputUnitD0Ev+00000014
[ 13] 0x000000014b8ec7fe /Users/DanielePucci/src/install/lib/libYARP_OS.1.dylib+00436222 _ZN4yarp2os4impl8PortCore10cleanUnitsEb+00000414
[ 14] 0x000000014b8e9686 /Users/DanielePucci/src/install/lib/libYARP_OS.1.dylib+00423558 _ZN4yarp2os4impl8PortCore9reapUnitsEv+00000470
[ 15] 0x000000014b8e902e /Users/DanielePucci/src/install/lib/libYARP_OS.1.dylib+00421934 _ZN4yarp2os4impl8PortCore3runEv+00001006
[ 16] 0x000000014b94e337 /Users/DanielePucci/src/install/lib/libYARP_OS.1.dylib+00836407 _Z18theExecutiveBranchPv+00000199
[ 17] 0x000000014b94f6b1 /Users/DanielePucci/src/install/lib/libYARP_OS.1.dylib+00841393 _ZNSt3__114__thread_proxyINS_5tupleIJPFjPvES2_EEEEES2_S2_+00000097
[ 18] 0x00007fff85ec1c13            /usr/lib/system/libsystem_pthread.dylib+00015379 _pthread_body+00000131
[ 19] 0x00007fff85ec1b90            /usr/lib/system/libsystem_pthread.dylib+00015248 _pthread_body+00000000
[ 20] 0x00007fff85ebf375            /usr/lib/system/libsystem_pthread.dylib+00004981 thread_start+00000013

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.
~~~

traversaro, 2016-01-21:
It this problem reproducible?

traversaro, 2016-02-14:
Ping.

DanielePucci, 2016-04-19:
Sorry @traversaro, did not notice this ping. I did not understand how it can be reproduced. It has just happened now.

traversaro, 2016-04-19:
From the stacktrace could be some weird race condition in `PortCoreInputUnit`, I think it could be related to https://github.com/robotology/yarp/issues/621 . If some good will man want to replicate it, he can just compile YARP with test support and then run in the YARP build directory this command
~~~
ctest -R PortTest -VV ---repeat-until-fail 10000 --output-on-failure
~~~
That will run the test until it fails.

traversaro, 2016-04-19:
cc @drdanz

traversaro, 2016-04-23:
Disclaimer: `repeat-until-fail` is available only since CMake 3.3 .

francesco-romano, 2016-12-15:
As it seems related to yarp and because it happens so rarely we can close this for now

closed 2016-12-15 by francesco-romano