robotology/wb-toolbox #15: Gazebo Synchroniser requires gazebo open even in compilation

francesco-romano, 2016-03-10:
`mdlStart` seems to be called also during compilation.
The clock port is opened there.

This forbids the possibility to have the gazebo synchroniser block even when we work on the robot.
We can open the port in the `mdlOutputs`  function or understand if the start can be called only on the real run of the simulation.

traversaro, 2016-03-10:
By 
> have the gazebo synchroniser
you mean "have it in a model" or have it in the block library?

francesco-romano, 2016-03-10:
Model.
A subsystem can be disabled. In this case it is not considered while the simulation runs.
But, of course, it must be compiled and initialized.

traversaro, 2016-03-10:
I see.

francesco-romano, 2016-12-06:
@S-Dafarra was complaining that this is not solved.
Reopen for now.
As soon as I will be able to have gazebo back, I will test it on my computer

closed 2016-12-13 by francesco-romano