robotology/wb-toolbox #16: Properly (and consistently) format the input, output and names in blocks

francesco-romano, 2016-03-14:
e.g. Base Pose => w^H_b

francesco-romano, 2016-03-14:
@traversaro with the reactions to comments, GitHub is now really the Facebook of developers [cit.] :smile:

MichaelBombile, 2017-04-18:
Hi , @francesco-romano @gabrielenava we are trying to implement a reactive walking controller on iCub2 at EPFL LASA, and to do so we are trying to combine it with the matlab torqueBalancing module by exploiting the ComDes:i and the qDes:i input ports. However in our implementation we tried to add an inverse Kinematics box from the WBT_toolbox in order to compute the qDes and it keeps on giving the following error msg: Error:Error reported by S-function 'WBToolbox' in 'torqueBalancingAndWalking/Inverse Kinematics1/S-Function':

[mdlStart] Cannot retrieve string from robot part parameter.

Here are the configuration parameters during the compilation:

[INFO][robotStatus::robotConfig] Configuring...
[INFO][robotStatus::robotConfig] wbiConfFileFullPath is: /home/michael/icub/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/yarpWholeBodyInterface.ini
[INFO][robotStatus::robotConfig] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][robotStatus::robotConfig] robot_fixed is false!
[DEBUG][robotStatus::robotConfig] After reading from config file, params are
[DEBUG][robotStatus::robotConfig] robot name: icubGazeboSim
[DEBUG][robotStatus::robotConfig] local name: matlabTorqueBalancing
[DEBUG][robotStatus::robotConfig] world reference frame: l_sole
[DEBUG][robotStatus::robotConfig] WBI ID List: ROBOT_TORQUE_CONTROL_JOINTS_WITHOUT_PRONOSUP

Please, can anyone tell us what is wrong and how we can fix it?

Thanks in advance.

francesco-romano, 2017-04-19:
Hi @MichaelBombile 
Can you check now?

BTW: please don't post in unrelated issues, please open new issues!
Also, avoid to spam your requests in all the robotology repositories.

Last thing: from your output I think you have also the old toolbox installed. I don't know what may happen if both are present.

MichaelBombile, 2017-04-19:
Hi @francesco-romano,
Thanks for your response.  
In fact, I installed the toolbox through the codyco-superbuild. I will update it, test it and I will let you know.