robotology/wb-toolbox #17: Take gravity direction as a configurable option

DanielePucci, 2016-03-18:
The gravity direction used for computing the generalised bias forces is set internally, i.e. 

https://github.com/robotology/WB-Toolbox/blob/master/toolbox/src/InverseDynamics.cpp#L177

It would be desirable to get it as a configurable option.

@traversaro @nunoguedelha @francesco-romano

nunoguedelha, 2016-03-18:
Maybe that's what you meant, but I think that it would be better if it was retrieved from the IMU measurement, to match Gazebo parameter.

traversaro, 2016-03-18:
@nunoguedelha The "model" block are not supposed to read data directly , but they provide a flexible input structure such that the user can choose if hardcode the gravity or read it from the IMU.

nunoguedelha, 2016-03-18:
ok, sure. Thanks

francesco-romano, 2016-03-22:
Volunteers for testing?

francesco-romano, 2016-03-24:
Closed by https://github.com/robotology/WB-Toolbox/pull/18

closed 2016-03-24 by francesco-romano