robotology/wb-toolbox #20: Centurial Momentum Matrix

DanielePucci, 2016-04-07:
It would be very good if the toolbox could provide the users with the centroidal momentum matrix.

azadm, 2016-04-11:
I strongly agree on this

DanielePucci, 2016-04-11:
Hi @azadm, for the time being, when I have to evaluate the CMM, I basically do two things:

1. often I have to evaluate terms of the kind: CMM(q)v, and to do this, it suffices to call the function H(q,nu) = CMM(q)nu  with the right arguments;

2. You can build the matrix CMM by columns. To do this, call the function H(q,nu) = CMM(q)nu a number of times equal to the degrees of freedom you are working with. In particular, let c_i denote the ith column of the CMM. Then, one has c_i = CMM(q)e_i, where e_i = [0 0 . . 1 . . 0], where the "1" is at the ith position.

azadm, 2016-04-12:
Thanks @DanielePucci

diegoferigo, 2017-11-13:
The block `Centroidal Momentum` returns the matrix as a signal. See ef08455

traversaro, 2017-11-13:
The Centroidal Momentum Matrix is the "jacobian" of the centroidal momentum w.r.t. the robot velocity, and it is not implemented by commit https://github.com/robotology/WB-Toolbox/commit/ef084555f097eb63aaef87febbc6ebfb2ed5943e .

traversaro, 2017-11-13:
See http://www.ambarish.com/paper/20080707_Orin_Goswami_momentum_matrix_IROS2008.pdf and http://ai2-s2-pdfs.s3.amazonaws.com/c854/a692d56b5c1f3b3d0eb61ed4ac47b30c47cc.pdf for more details about the CMM.

francesco-romano, 2017-11-13:
Sorry, my fault

traversaro, 2019-04-26:
@DanielePucci just to let you know,  implementing this will probably require a bit of modifications in KinDynComputatuins and on the WB-Toolbox side, but nothing too difficult. It would be a perfect task for someone that needs to get in touch in a "vertical" way with the WB-Toolbox/BlockFactory/iDynTree "stack".

gabrielenava, 2019-04-29:
Note that in `whole-body-controllers` there is an high-level block I've implemented called [CMM_iCub_23_25_DoFs](https://github.com/robotology/whole-body-controllers/tree/master/library/simulink-library) that allows to retrieve the CMM for the 23/25 DoFs iCub (number of DoFs is hard-coded). It implements the procedure suggested in [this comment](https://github.com/robotology/wb-toolbox/issues/20#issuecomment-208432218).

traversaro, 2020-06-12:
At the iDynTree level, it probably is sufficient to just add a `KinDynComputations::getCentroidalMomentumMatrix` method
As there are already a: 
* `getCentroidalTotalMomentum` method, that returns the centroidal momentum ${}_{G[A]} \mathrm{h} \mathbb{R}^6$  (when the velocity convention is mixed) 
* `getLinearAngularMomentum` method, that returns the momentum ${}\_{B[A]} \mathrm{h} \in \mathbb{R}^6$  (when the velocity convention is mixed) 
* `getLinearAngularMomentumJacobian` method, that returns the matrix $J\_{MomJac} \in \mathbb{R}^{6 \times (n+6}$ such that ${}_{B[A]} \mathrm{h} = J\_{MomJac} \nu$  (when the velocity convention is mixed) 

We just need to be implement a `getCentroidalTotalMomentum` or `getCentroidalMomentumMatrix` that returns the matrix $J_{CMM} \in \mathbb{R}^{6 \times (n+6}$ such that ${}_{G[A]} \mathrm{h} = J_{CMM} \nu$  (when the velocity convention is mixed)

traversaro, 2020-06-12:
At the WB-Toolbox level, once an appropriate method is available in the `iDynTree::KinDynComputations` class, it is sufficient to check how a similar block has been implemented, for example in https://github.com/robotology/wb-toolbox/blob/master/toolbox/library/src/Jacobian.cpp .

GiulioRomualdi, 2020-06-26:
I was thinking to open a PR on iDynTree regarding this. I've already the code almost ready. Let me know if there is already planning to work on this

traversaro, 2020-06-26:
Not on my side, and I suspect no one else is working on this.

GiulioRomualdi, 2020-06-27:
https://github.com/robotology/idyntree/pull/706 implements the CMM in iDynTree.

DanielePucci, 2020-06-27:
Good job @GiulioRomualdi, this is something we should have done a long time ago. Thanks a lot

CC
@robotology/iit-dynamic-interaction-control

traversaro, 2020-06-29:
https://github.com/robotology/idyntree/pull/706 has been merged.

GiulioRomualdi, 2020-06-29:
> robotology/idyntree#706 has been merged.

Now I think a Simulink block is required to use the function in wb-toolbox

DanielePucci, 2020-06-29:
CC
@robotology/iit-dynamic-interaction-control

CarlottaSartore, 2020-07-03:
> Now I think a Simulink block is required to use the function in wb-toolbox

@GiulioRomualdi  is there any plan about this? I need the Simulink block so if there is no one assigned to it yet I can take care of it (I will need a little help since it is the first time that I implement something like this  )

traversaro, 2020-07-03:
> @GiulioRomualdi is there any plan about this? I need the Simulink block so if there is no one assigned to it yet I can take care of it (I will need a little help since it is the first time that I [...]

If I remember correctly @GiulioRomualdi  is not going to work on it (@GiulioRomualdi  feel free to correct me). @CarlottaSartore if you need a starting point, I would start copying the [boilerplate](https://en.wikipedia.org/wiki/Boilerplate_code) from the Jacobian block, that has many similary with the CentroidalMomentumMatrix computation. You can find the code in: 
* https://github.com/robotology/wb-toolbox/blob/master/toolbox/library/src/Jacobian.cpp
* https://github.com/robotology/wb-toolbox/blob/master/toolbox/library/include/WBToolbox/Block/Jacobian.h

The main difference is that probably for CMM you don't need the frame parameter. You will then need to add the block themself in the simulink library, that you can find in https://github.com/robotology/wb-toolbox/blob/master/matlab/library/WBToolboxLibrary_repository.mdl . For that, the guide in https://robotology.github.io/blockfactory/mkdocs/create_new_library/ can be a useful  starting point (even if in this case you don't need to create a new library, just a new block in an existing directory.

CarlottaSartore, 2020-07-03:
Thank you really much @traversaro , I will start from the material you provided !

GiulioRomualdi, 2023-09-27:
Given https://github.com/robotology/wb-toolbox/pull/194 I think we can close this issues

traversaro, 2023-09-27:
Fixed by @CarlottaSartore in https://github.com/robotology/wb-toolbox/pull/194 .

closed 2023-09-27 by traversaro