robotology/wb-toolbox #26: Synchronisation problems between Gazebo and estimation threads

DanielePucci, 2016-06-21:
**Description:**
The Gazebo simulation and the estimation threads in charge of estimating joint position/velocities/etc are not synchronised even when using the appropriate block, and this introduces an undesired machine-dependence of the controllers performances.

@traversaro @francesco-romano 
_____

**Further Information**:
- *System Information:* `Darwin Danieles-MacBook-Pro.local 15.5.0 Darwin Kernel Version 15.5.0: Tue Apr 19 18:36:36 PDT 2016; root:xnu-3248.50.21~8/RELEASE_X86_64 x86_64`
- *Working branch:* `master`
- *Current commit:* `cc53fb3c750899a0dddcae413a9484a6dfee24b4`
- *YARP ROOT:* `/Users/DanielePucci/src/yarp`
- *ICUB_ROOT:* `/Users/DanielePucci/src/icub-main`
- *YARP_DATA_DIRS:* `/Users/DanielePucci/src/install/share/yarp:/Users/DanielePucci/src/install/share/iCub:/Users/DanielePucci/src/codyco-superbuild/build/install/share/codyco:/Users/DanielePucci/src/install/share/ICUBcontrib`
- *YARP_ROBOT_NAME:* `iCubGenova05`
- *Yarp Installation:* `/Users/DanielePucci/src/install/bin/yarpserver`
- *LD_LIBRARY_PATH:*``
- *LD_PRELOAD:*``
- *DYLD_INSERT_LIBRARIES:*``

traversaro, 2016-06-21:
The `/clock` port is still published by the `gazebo_yarp_clock` plugin, even if the simulation is paused and stepped by the simulink gazebo synchronization block. 
Starting MATLAB prepending the `matlab` command with `YARP_CLOCK=/clock` should solve the issue, however I don't know if there is any issue with that with our current workflow.
@DanielePucci suggested using `setenv` to set the `YARP_CLOCK` enviromental variable, however it should investigated if that is sufficent, as the `YARP_CLOCK` enviromental variable is readed only in the yarp init function : https://github.com/robotology/yarp/blob/0e862d8daec639d74d94f79460f7c21aff619072/src/libYARP_OS/src/Network.cpp#L620 .

traversaro, 2016-06-21:
cc @gabrielenava @S-Dafarra

Yeshasvitvs, 2017-05-18:
@diegoferigo FYI

francesco-romano, 2017-07-19:
https://github.com/robotology/WB-Toolbox/issues/46 should fix this issue

diegoferigo, 2017-07-19:
@DanielePucci The ImpedanceControl didn't have `YARP_CLOCK` set. Exporting this variable will affect the filtering of the speed signal we talked about f2f, and it is an alternative solution of setting the `readSpeedAccFromControlBoard` variable in `yarpWholeBodyInterface.ini`.

DanielePucci, 2017-07-19:
Mmmm it seems weird to me since I'm pretty sure that `torqueBalancing` back in time did have the synchroniser correctly set.

diegoferigo, 2017-07-19:
The synchronizer is a different thing as far as I understood. Despite processes are spawned by simulink, which controls gazebo stepping through the syncronizer you mentioned, they run by default using the system clock. In order to sync them with simulink, using `YARP_CLOCK` is required. It is a complicated loop, but in this way `YARP_CLOCK` is taken from gazebo, which is set step after step as said by simulink.

traversaro, 2018-12-22:
I think this was solved in https://github.com/robotology/wb-toolbox/pull/56 .

diegoferigo, 2019-01-03:
PR #56 removed the abstraction layer provided by the `whole-body-interface`. In its `yarp` implementation we were filtering the estimated quantities with tunable low-pass filters.

Right now the measurements provided by the _GetMeasurement_ block are the raw signals, and it is user's responsibility to filter them properly with the _DiscreteFilter_ introduced in #47.

This does not mean that the `YARP_CLOCK` env variable should not be set, we still need it in order to set the network clock as the default clock. I remember that during the last year the clock functionality in YARP has been rewritten, I'm not sure if I am missing something. @traversaro

diegoferigo, 2019-01-03:
I think this issue can be closed since the solution is to use the `YARP_CLOCK` env variable. In order to avoid it, we should go towards #164.

closed 2019-01-03 by diegoferigo