robotology/wb-toolbox #30: Whole body control with a simultaneous use of more than one type of controller

TeoKa, 2016-07-15:
iCub is balancing on two feet. Position (direct) based controller is used to perform a walking trajectory. At a certain time i want to switch a single joint controller, pitch ankle joint, from position to impedance before the step-down: to adapt a foot to a possible uneven surfsce, while keeping all other controllers to position control.
Any idea if it is possible to accomplish this switch? WB-toolbox block(s) suggested?

Thanks in advance for your help
@traversaro @francesco-romano

francesco-romano, 2016-07-16:
First implementation done in https://github.com/robotology/WB-Toolbox/commit/777450ae1b1dd977a19e3f5d361dce13f94fdaa4

Please @TeoKa check if this is ok for your need (and of course if it works..)

TeoKa, 2016-07-22:
It works, thanks

closed 2016-07-22 by TeoKa