robotology/wb-toolbox #32: Minimum Jerk generator and variable settling time

francesco-romano, 2016-07-25:
The minimum jerk trajectory block (which uses the `iCub::ctrl::minJerkTrajGen` class) has now the option to change the settling time during the simulation.

The behaviour of the "position" can be observed in the following plot, where the reference is a step of amplitude 1 at t=2s, and the settling time is changed at t = 2.75s.

[image]

We expect the signal to have continuous derivatives, but if we plot the first derivative we obtain the following:

[image]
where the first plot is the position signal, and the second plot has the derivative output by the generator in yellow and the real derivative in blue.

This issue is (probably) just for reference as (maybe) the `minJerkTrajGen` was never thought to be used in this way. 

cc @DanielePucci @traversaro @S-Dafarra 
@pattacini

pattacini, 2016-07-25:
Apparently, [this line](https://github.com/robotology/icub-main/blob/master/src/libraries/ctrlLib/src/minJerkCtrl.cpp#L441) (executed within the [**`computeCoeffs`**](https://github.com/robotology/icub-main/blob/master/src/libraries/ctrlLib/src/minJerkCtrl.cpp#L415) method, which in turn gets called upon the request for changing the settling time) is responsible for resetting the computation of the velocity. The comment therein says that it's done to avoid spikes. @andreadelprete - the original author - might add some further comments on this.

However, it's strange that the velocity remains clamped to zero afterwards. I don't see how this can happen, given that [**`computeNextValues`**](https://github.com/robotology/icub-main/blob/master/src/libraries/ctrlLib/src/minJerkCtrl.cpp#L455) handles both position and velocity filters indifferently.

francesco-romano, 2016-07-25:
Thanks @pattacini.
So this was not the intended behaviour. 

Btw: sorry @pattacini it does not remain zero but the plot does not show it.
The following plot is a zoom:

[image]

pattacini, 2016-07-25:
Yep, I think the original purpose was to allow tuning settling time only in stationary conditions.

francesco-romano, 2016-07-25:
Ok.
I'll keep this open until I find a nice way to include it in a sort of f.a.q. of the project.

Thanks @pattacini

andreadelprete, 2016-07-26:
Hi guys, 

it's been a long time since I wrote this class, so I don't remember exactly the details of the math behind it. Maybe you could try replacing the line pointed out by @pattacini :
` velFilter->init(zeros(dim), pos);`
with:
` velFilter->init(vel, pos);`
Same thing for the acceleration. Not sure whether it will work, but it's worth a try.

pattacini, 2016-07-26:
@andreadelprete hope you're doing well :smile:

closed 2016-12-15 by francesco-romano