robotology/wb-toolbox #33: Block [15-DoFs WBI to YARP] indeces are out of range

hu-yue, 2016-08-26:
**Description:**

I haven't used the toolbox since ages, I wanted to use it again today, but I spotted a problem in the `15-DoFs WBI to YARP` block.
I am assuming that this block is supposed to be used with a robot with reduced number of DOF (`iCubHeidelberg`), and that therefore the indeces of the joints should go from 1 to 15.
Instead in the legs the indeces are `12:17` and `18:23`, which are of course out of range if used with `iCubHeidelberg`.
As a side note, also in the block description you still say `23` instead of 15.

_____

**Further Information**:
- *System Information:* `Linux yue-ThinkPad-S5-S540 4.4.0-34-generic #53~14.04.1-Ubuntu SMP Wed Jul 27 16:56:40 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`
- *Working branch:* ``
- *Current commit:* ``
- *YARP ROOT:* `/home/yue/iCubSW//yarp`
- *ICUB_ROOT:* `/home/yue/iCubSW//icub-main`
- *YARP_DATA_DIRS:* `/home/yue/iCubSW//yarp/build/share/yarp:/home/yue/iCubSW//icub-main/build/share/iCub/:/home/yue/iCubSW/icub-tests-Heidelberg01/suits:/home/yue/iCubSW/codyco-superbuild/build/install/share/codyco`
- *YARP_ROBOT_NAME:* `iCubHeidelberg01`
- *Yarp Installation:* `/usr/local/bin/yarpserver`
- *LD_LIBRARY_PATH:*``
- *LD_PRELOAD:*``
- *DYLD_INSERT_LIBRARIES:*``

francesco-romano, 2016-08-26:
Hi @hu-yue .
These kind of blocks were directly ported from the old toolbox, so I did not modify them when I developed the WB-Toolbox.
I'd always wanted to make this kind of block generic, but at the current state I've never managed to.
I'll take a look at the current block so to fix it.

francesco-romano, 2016-08-26:
@hu-yue can you check if it is all right now, and reopen the issue otherwise?

closed 2016-08-26 by francesco-romano

hu-yue, 2016-08-26:
@francesco-romano it works fine, thanks =)