robotology/wb-toolbox #35: Detect pressing of fault button

francesco-romano, 2016-11-10:
It would be nice to be able to detect the fault button so as to stop the controller or to better understand the scopes.

Brainstorming with @randaz81 we come out with two possible heuristics (unfortunately there is not a clean way to retrieve this status from the robot)

1. monitor the PWM to detect when they are all zero (I tested on one part of the robot and unfortunately this was not reliable.)
2. Check if all the joints are in idle and in this case try to set a control mode on one joint and check (with a get after a timeout of at least 10ms) if the joint has changed control mode. Unfortunately this modifies also the state of the robot and so it is not so clean as solution.

S-Dafarra, 2016-11-11:
A possible workaround is to read the ``/controlBoardDumper/left_leg/getMotorsPwm`` from simulink with the following scheme:
[image]

traversaro, 2018-12-22:
The more I think about this, the more I get the impression that the robot software infrastructure (iCub) itself should provide a way to clearly expose the state of the emergency button. The firmware have this information in some way ( https://github.com/robotology/icub-firmware/search?q=external+fault&unscoped_q=external+fault ), so we should find a proper way to expose it, either using an existing YARP device interface, or something similar.