robotology/wb-toolbox #39: [Announcement] Related publication

francesco-romano, 2017-01-05:
Dear all (especially @robotology/codyco-developers)

we now have an associated (conference paper) publication with the Whole-Body interface architecture (including this toolbox).
Please cite the following publication if you are using WB-Toolbox for your own research and/or robot controllers

> Romano, F., Traversaro, S., Pucci, D., Del Prete, A., Eljaik, J., Nori, F. "A Whole-Body Software Abstraction layer for Control Design of free-floating Mechanical Systems", 1st IEEE International Conference on Robotic Computing, 2017.

francesco-romano, 2017-01-05:
Preprint available at https://arxiv.org/abs/1701.00935

closed 2017-01-05 by francesco-romano

iron76, 2017-01-09:
Full citation:

> Francesco Romano, Silvio Traversaro, Daniele Pucci, Jorhabib Eljaik, Andrea Del Prete and Francesco Nori. A Whole-Body Software Abstraction layer for Control Design of free-floating Mechanical Systems. IEEE Robotic Computing 2017.

francesco-romano, 2017-01-16:
```
@INPROCEEDINGS{RomanoWBI17, 
author={F. Romano and S. Traversaro and D. Pucci and A. Del Prete and J. Eljaik and F. Nori}, 
booktitle={2017 IEEE 1st International Conference on Robotic Computing}, 
title={A Whole-Body Software Abstraction layer for Control Design of free-floating Mechanical Systems}, 
year={2017}, 
}

```