robotology/wb-toolbox #40: Simulink compilation stops at "Compiling: evaluating block parameters: Started"

DanielePucci, 2017-01-05:
**Description:**
After updating at the latest `master` branches of `yarp`, `icub-main`, `gazebo-yarp-plugins`, and `codyco-superbuild`, it seems to me that when compiling the `torqueBalancing`, Simulink compilation gets stuck at the stage `Compiling: evaluating block parameters: Started`.

@francesco-romano @traversaro @fjandrad
_____

**Further Information**:
- *System Information:* `Darwin Danieles-MacBook-Pro.local 15.6.0 Darwin Kernel Version 15.6.0: Wed Nov  2 20:30:56 PDT 2016; root:xnu-3248.60.11.1.2~2/RELEASE_X86_64 x86_64`
- *Working branch:* `master`
- *Current commit:* `3223ec8d249eadddce01bc9f56e7a9e9252a026e`
- *YARP ROOT:* `/Users/DanielePucci/src/yarp`
- *ICUB_ROOT:* `/Users/DanielePucci/src/icub-main`
- *YARP_DATA_DIRS:* `/Users/DanielePucci/src/install/share/yarp:/Users/DanielePucci/src/install/share/iCub:/Users/DanielePucci/src/codyco-superbuild/build/install/share/codyco:/Users/DanielePucci/src/install/share/ICUBcontrib`
- *YARP_ROBOT_NAME:* `iCubGenova02`
- *Yarp Installation:* `/Users/DanielePucci/src/install/bin/yarpserver`
- *LD_LIBRARY_PATH:*``
- *LD_PRELOAD:*``
- *DYLD_INSERT_LIBRARIES:*``

fjandrad, 2017-01-05:
I have a previous version of Yarp, icub-main and codyco-superbuild.  WBIToolboxControllers is up to date in Master branch. The torqueBalancing2012b.mdl from  WBIToolboxControllers compiles for me without problems.

My version on Yarp is:
commit a265000533559630afe770b4c48a6a7bf8892324
Merge: 259f58b bbea462
Author: Ugo Pattacini <ugo.pattacini@iit.it>
Date:   Thu Sep 8 12:05:13 2016 +0200

My version on icub-main is:
commit 88377ab826d5bbe9cd3880d65685aae719a40ac7
Author: Alessandro Scalzo <alessandro.scalzo@iit.it>
Date:   Thu Sep 8 15:45:52 2016 +0200

My codyco-superbuild version is:
commit 6af140926b86edf378f02cb2b0c1dbbdf70e92e0
Author: Silvio Traversaro <pegua1@gmail.com>
Date:   Fri Aug 26 20:06:21 2016 +0200

And my WBIToolboxControllers version is:
commit 15d3158c047ec2aa207930519175a4a4ad27ac98
Author: FirstName <daniele.pucci85@gmail.com>
Date:   Fri Dec 16 11:28:01 2016 +0100

fjandrad, 2017-01-05:
I updated Yarp through the following steps:

git pull
cd build
ccmake ..
make -j4
sudo make install

Now matlab crashes when compiling and I get the following message in the terminal where I opened matlab

Segmentation fault (core dumped)

traversaro, 2017-01-05:
There is a [ABI mismatch](https://en.wikipedia.org/wiki/Application_binary_interface)  between `yarp`, `icub-main` and all the libraries in `codyco-superbuild` that depend on them. To understand the cause of the problem, you can try to keep `yarp` and `icub-main` to the ancient commits, and first update just the `codyco-superbuild`, and compile it with the old `yarp` and `icub-main` commits. Otherwise we can try to keep the new yarp, and try to compile  the old `icub-main` and `codyco-superbuild`, but I think it would not work because the old `icub-main` is not compatible anymore with the new `yarp`.

DanielePucci, 2017-01-05:
Guys, I re-updated, re-builded, and cleaned the caches. Things started working. I close this issue then.
@francesco-romano Everything was at master branch.

DanielePucci, 2017-01-05:
P.S. I've also erased the `splrj` Simulink build folder in the second round.

closed 2017-01-05 by DanielePucci