robotology/wb-toolbox #43: Add automatic tests to the repository

francesco-romano, 2017-03-17:
Probably we can also create the simulink models needed for tests with matlab scripts

diegoferigo, 2017-11-21:
For testing the `WB-Toolbox 3.0` I'm writing `mdl` models containing simple tests for elementary blocks. If these models run without any error, it means they compile correctly, but the signals cannot be compared in such a way to make it compatible with Continuous Integration tools.

A possible idea might be to use Matlab scripts to configure the model, and exporting signals to the WS from the simulation. This might be useful also for comparing the outputs of blocks which under the hood use `iDynTree` with the iDynTree Matlab's bindings.

On the long run, I'd like to explore two ways for improving these tests:

* If the Matlab license allows it, the right combination of Docker image + encrypted Travis variables might provide a CI toolchain
* Waiting that the automatic code generation works, and test it directly from C++ side (probably the best option)

traversaro, 2017-11-21:
A test for a WB-Toolbox based controller is available in https://github.com/vvv-school/assignment_computed-torque-simulink/tree/master/smoke-test .

diegoferigo, 2017-11-21:
@traversaro Awesome, thanks! Basically the scaffold for the first option is already there. With some docker magic, porting it to Travis should not take so long.

However, I still prefer proceeding at least initially with the second option considering the plans about the automatic code generation. We would need in any case tests on the generated code, hence I'd rather go for killing two birds with one stone.

diegoferigo, 2018-02-24:
In the next release of `WB-Toolbox`, the library will be separated from simulink and matlab libraries. This new `WBToolboxLibrary` will contain the blocks that can be therefore tested in Travis.

The way to test the functionality of a single block is not yet clear. We can think of a direct test of the implemented functionality (e.g. for iDyntree-related blocks, checking how iDyntree checks the validity of the `KinDynComputations` methods), or for more complex scenarios by creating a Simulink model and testing the generated C++ code from it.

cc @traversaro @francesco-romano

francesco-romano, 2018-02-26:
Sorry, I do not understand what you would like to test.
If it is possible to autogenerate code programmatically (i.e. without MATLAB Gui), then I would suggest to test both the Simulink model (writing tests in simulink) and in C++ for the generated code.
I know that Simulink is not available in travis, but I still think that being this a simulink library, the most important part to be tested is the simulink integration.