robotology/wb-toolbox #46: WB-Toolbox 3.0

francesco-romano, 2017-07-19:
The idea is to migrate away from `yarp-wholebodyinterface`.
The model will use directly iDynTree (in particular `KinDynComputations`).
Reading sensors and commanding actuators can be done through the YARP `ControlBoardRemapper` control board.

The tricky part remains in moving the estimators and filters away from the state interface.
The filter classes can be exposed as blocks.

traversaro, 2017-07-19:
Possible challenges: 
 * WB-Toolbox stateless model blocks vs `iDynTree::KinDynComputations` stateful interface 
 * Degrees used in YARP interfaces vs Radians used in WB-Toolbox and iDynTree interfaces

diegoferigo, 2017-08-10:
@francesco-romano I edited your first post adding a draft of the steps required for the migration. Feel free to add more details, I'll keep it updated while working on it.

closed 2018-01-25 by diegoferigo