robotology/wb-toolbox #57: Compile error when using setReferences with sublist

S-Dafarra, 2017-09-22:
I tried to use ``setReferences`` with a sublist composed of only ``'r_knee'`` but I got the following compile error from simulink:
``
S-function method mdlSetInputPortDimensionInfo cannot change the dimension of input port 1 of 'icubDump/Set References/S-Function' once it has been set
Component:Simulink | Category:Modelerror
Error in port widths or dimensions. Output port 1 of 'icubDump/Set References/References' is a one dimensional vector with 1 elements.
``
@diegoferigo @francesco-romano

diegoferigo, 2017-09-25:
I quickly tried to debug this error to let @S-Dafarra have a working block. Referencing:

https://github.com/robotology/WB-Toolbox/blob/3f8895ab20197c6d05821b8ddd4bf4607dafd363/toolbox/src/SetReferences.cpp#L28-L51

it looks like this code tries to read as "boolean data" the second argument, which at the current state is the label of the dropdown menu of the mask (`full` or `reduced list` if I recall well).

I temporary fixed his block creating inside the mask a new boolean (logic) variable `isFullControl`, which value is set depending on the dropdown menu status, and pass it as second parameter to the S-Function. Then, a working joint list was in the `'(r_knee r_ankle)'` format.
@francesco-romano

francesco-romano, 2017-09-25:
This is very strange, as I am sure I tested it and it was working (we also used for some experiments).
So something has changed in the meanwhile.
I will debug this ASAP

francesco-romano, 2017-09-25:
The problem was when I abstracted the toolbox implementation from Simulink.

The following line
```cpp
m_fullControl = static_cast<int>(mxGetScalar(ssGetSFcnParam(S,WBIBlock::numberOfParameters() + 2))) == 1;
```
was converted to a boolean, while I actually meant to compare with `1`.

closed 2017-09-25 by francesco-romano