robotology/wb-toolbox #58: Clarification on how Holder block works

gabrielenava, 2017-10-26:
I did the following test: a ramp signal (starting ramp value: 10) is sent into a subsystem, containing an HOLDER block. Every 2 seconds, the subsystem containing the HOLDER is enabled/disabled for a period of 1s each. See pictures attached.

[image]

Inside the subsystem:

[image]

These are the output: (Simulation time: 30s)

- Signal sent to the ENABLE block:

[image]

- Ramp signal (no HOLDER)

[image]

-  **Ramp signal passing through the HOLDER block**

[image]

Now we know that the output of HOLDER block **will always be the first value sent to the block**. Maybe this may be an issue for some applications.

francesco-romano, 2017-10-26:
Thanks @gabrielenava 
I think that it may be worthwhile to move the implementation of the Holder block to c++ (the implementation is trivial).

Currently blocked by #46

S-Dafarra, 2017-10-26:
This is not a problem of Holder block, but rather it is an intended behavior of the ``Enable`` Simulink block. 
If you select ``reset`` from the drop menu in the options of the ``Enable`` block, the internal state of the ``Holder`` block is reset when enabled, thus changing the value of the ``Holder`` with the new one.
[image]

francesco-romano, 2017-10-26:
Thanks @S-Dafarra.
I knew about that option, but I don't know if @gabrielenava tried it. @gabrielenava can you try if the holder block still exhibits the same problem with this option changed?

gabrielenava, 2017-10-26:
No it doesn't. I didn't know about this option, changing it solves the issue.

francesco-romano, 2017-10-26:
Great. For the time being this can remain as simple matlab code if it is compatible with code generation

closed 2017-10-26 by francesco-romano