robotology/wb-toolbox #66: Test backwards compatibility of WB-Toolbox 3.0

diegoferigo, 2017-12-07:
PR https://github.com/robotology/WB-Toolbox/pull/56 was tested on the following system:

* Ubuntu 17.04 (Zesty Zapus)
* Matlab 9.2 (2017a), Simulink (8.9)
* gcc version 6.3.0 20170406 (Ubuntu 6.3.0-12ubuntu2)
* clang version 4.0.0-1ubuntu1 (tags/RELEASE_400/rc1)

Here below a list of issues found during the tests.

## Matlab

### 1. Properties size check of Matlab classes

- [ ] [WBToolboxConfig.m](https://github.com/diegoferigo/WB-Toolbox/blob/47f3b5a83f8f47214df0aa8c75174cac382324fb/toolbox/matlab/%2BWBToolbox/WBToolboxConfig.m#L10-L24)
- [ ] [WBTPIDConfig.m](https://github.com/diegoferigo/WB-Toolbox/blob/47f3b5a83f8f47214df0aa8c75174cac382324fb/toolbox/matlab/%2BWBToolbox/WBTPIDConfig.m#L20-L23)
- [ ] [PID.m](https://github.com/diegoferigo/WB-Toolbox/blob/47f3b5a83f8f47214df0aa8c75174cac382324fb/toolbox/matlab/%2BWBToolbox/PID.m#L18-L21)

### 2. Apices

- [ ] [WBToolboxConfig](https://github.com/diegoferigo/WB-Toolbox/blob/47f3b5a83f8f47214df0aa8c75174cac382324fb/toolbox/matlab/%2BWBToolbox/WBToolboxConfig.m#L119): from `"'"` to `''''`

## C++

* [Curly braket tuple initialization](https://github.com/diegoferigo/WB-Toolbox/blob/47f3b5a83f8f47214df0aa8c75174cac382324fb/toolbox/src/SetLowLevelPID.cpp#L123) fails on gcc 5.4 / Ubuntu 16.04

traversaro, 2018-12-22:
As whole-body-controllers have been ported successfully, I guess this was solved, right @diegoferigo ?

diegoferigo, 2019-01-03:
Yes, closing. It would have been nice testing it on other Ubuntu versions and maybe Windows, but let's do it with the most recent releases if needed.

closed 2019-01-03 by diegoferigo