robotology/wb-toolbox #68: Yarp port timeout

diegoferigo, 2017-12-19:
Using the _YarpRead_ block in **Blocking Read** configuration, if no data is received the whole Simulink (and Matlab) get stuck forever and they must be manually killed.

A timeout capability should be added, and this should generate an error. In this way the simulation would stop smoothly.

@S-Dafarra

diegoferigo, 2017-12-19:
@S-Dafarra Temporary branch based on the current `master`:

https://github.com/diegoferigo/WB-Toolbox/tree/yarpReadPortTimeout

S-Dafarra, 2017-12-19:
Thanks! I will test it right away!

S-Dafarra, 2017-12-19:
It seems to be working as expected!

diegoferigo, 2018-02-12:
Fixed by #74. @S-Dafarra Let me know if you face any issue. Now you can specify the timeout from _Yarp read_'s mask.

closed 2018-02-12 by diegoferigo