robotology/wb-toolbox #70: The YARP to DOF converter should label the output ports

S-Dafarra, 2018-01-31:
This goes in the direction of reducing possible source of misunderstanding when connecting this block to a scope, especially when the user does not remember/know the control boards order.

diegoferigo, 2018-02-04:
This feature would require the block to access the closest _Configuration_ block and read the order or the ControBoards list. It is doable without a big effort.

However, I will wait a bit before implementing it because it is somehow related to changes related to the coming automatic code generation support and its impact on block's masks.

diegoferigo, 2019-01-17:
This block has been removed from the toolbox in PR https://github.com/robotology/wb-toolbox/pull/147 in favor of user logic directly in Simulink. Closing.

closed 2019-01-17 by diegoferigo