robotology/wb-toolbox #72: SetLowLevelPIDs block fails to load the provided configuration

diegoferigo, 2018-02-06:
I changed the name of the class containing the configuration to `WBToolbox.PIDConfiguration`, but the `Initialization` function of the mask still uses the old name. It should be renamed.

As a temporary workaround, manually creating the object that the blocks expects works:

```matlab
myPIDConfig = WBToolbox.PIDConfiguration();
% ... Add WBToolbox.PID content
WBTPIDConfigStruct = myPIDConfig.getSimulinkParameters();
```

cc @vibhoraggarwal

diegoferigo, 2018-02-12:
Fixed by #74.

vibhoraggarwal, 2018-05-28:
@diegoferigo When I tried to input the PID values using:
`pids = WBToolbox.PIDConfiguration;`
`pids.addPID(WBToolbox.PID('l_hip_pitch', -0.62, 0, 0));`

I get the following error:
[image]

diegoferigo, 2018-05-29:
@vibhoraggarwal Handling apices in masks is a pain, there was an error which is now fixed in the Simulink Library commit of PR #118. Let me know if it works fine.

vibhoraggarwal, 2018-05-30:
I tried to use the `set Low Level PIDs` block. It still does not work, and gives me the error, as appended at the end of this comment.
Although, this is the same type of error which was overcome by using a trick mentioned on the [first comment of this thread](https://github.com/robotology/wb-toolbox/issues/72#issue-294709437); but, when I use that now, Matlab also gets stuck.

```Invalid setting in 'SquareTorque2016b/DataCollectin/T/Set LowLevel PIDs1/S-Function' for parameter 'WBTPIDConfigStruct'.
Caused by:
Error evaluating parameter 'WBTPIDConfigStruct' in 'SquareTorque2016b/DataCollectin/T/Set LowLevel PIDs1/S-Function'
Undefined function or variable 'WBTPIDConfigStruct'.

diegoferigo, 2018-07-04:
PR https://github.com/robotology/wb-toolbox/pull/135 introduced a ne block for setting PIDs with a new logic. Closing.

closed 2018-07-04 by diegoferigo