robotology/wb-toolbox #73: Matlab 2017b and Ubuntu 17:10

diegoferigo, 2018-02-08:
Simulink errors are not displayed due to a mismatch of the `libatk1.0` library. Matlab's requires a previous version.

Right now the only solution would be to downgrade the library (anyone wants to dedicate some time to it?).

#### Resources

- https://forum.manjaro.org/t/solved-matlab-r2017b-add-ons-and-other-windows-no-effect/35505
- https://it.mathworks.com/matlabcentral/answers/309629-cannot-open-add-on-window-on-linux
- https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=libatk1.0&searchon=names

traversaro, 2018-02-08:
I would expect Matlab to ship its own version of `libatk1.0` to solve this (until you have a `.mex` that need to link `libatk1.0` : ), but fortunatly that is not our case, right? ). We should open a support request from MathWorks on this.

diegoferigo, 2018-02-08:
> I would expect Matlab to ship its own version of libatk1.0 to solve this (until you have a .mex that need to link libatk1.0 : ), but fortunatly that is not our case, right? ).

Not at all luckily. I would expect the same, but for the time being I rolled back my development image to 16:04.

diegoferigo, 2018-08-03:
On Ubuntu Bionic 18.04 and Matlab 2018a the problem is still present. Be aware that this Matlab version supports officially up to Ubuntu 17.10.

However, a solution to this problem is the following:

```bash
cd /usr/local/MATLAB/R2018a/cefclient/sys/os/glnxa64
sudo mkdir exclude
sudo mv libglib-2.0.so* exclude/
```

Sources:

- https://it.mathworks.com/matlabcentral/answers/364551-why-is-matlab-unable-to-run-the-matlabwindow-application-on-linux
- https://it.mathworks.com/matlabcentral/answers/412260-r2018a-matlabwebappserver-installation-on-ubuntu-18-04

closed 2018-08-03 by diegoferigo