robotology/wb-toolbox #75: Unit of 'Reference speed' measurement in 'Set references' block

vibhoraggarwal, 2018-02-21:
It is not clear on first sight that what is the unit of measurement for `reference speed` in the block. Mentioning of the unit in the text would be much convenient.
[image]

francesco-romano, 2018-02-21:
Just be careful that from what I understood [in the code](https://github.com/robotology/wb-toolbox/blob/WB3.0/toolbox/src/SetReferences.cpp#L140) the velocity is not converted from `rad/s` to `deg/s` making the toolbox not consistent (reference positions and velocities are in radians, but the velocity for the trajectory generator is in degrees per seconds).

diegoferigo, 2018-05-11:
Fixed in https://github.com/robotology/wb-toolbox/issues/110

closed 2018-05-11 by diegoferigo