robotology/wb-toolbox #93: GetControlReferences block

diegoferigo, 2018-02-27:
The unmerged PR #54 introduced the support of reading control references (that might be streamed from outside Simulink, e.g. from a C++ application).

This work was done for a past experiment, and due to the undergoing `WB-Toolbox` refactoring, the PR was not merged.

I open this issue to keep track of this feature if we want to refactor it on the new version (this would require getting the real implementation from the `yarpWholeBodyInterface`).

S-Dafarra, 2018-03-09:
Can I ask to have a bigger priority for this. I would really need it soon for debug purposes (i.e. I want to check eventual delays and if what I ask to the boards is what is used). Thanks :grin:

diegoferigo, 2018-03-10:
Ok, I will include this block in the coming `3.1` release.