robotology/wb-toolbox #98: Compilation problems

S-Dafarra, 2018-03-08:
While compiling the toolbox I had the following issue:
```
/home/sdafarra/Software/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp: In member function ‘bool wbt::RobotInterface::mapDoFs()’:
/home/sdafarra/Software/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:126:58: error: invalid initialization of non-const reference of type ‘yarp::os::ConstString&’ from an rvalue of type ‘yarp::os::ConstString’
             if (!iAxisInfoPtr->getAxisName(axis, axisName)) {
                                                          ^
In file included from /home/sdafarra/Software/robotology-superbuild/build/install/include/yarp/os/ContactStyle.h:10:0,
                 from /home/sdafarra/Software/robotology-superbuild/build/install/include/yarp/os/Network.h:10,
                 from /home/sdafarra/Software/robotology-superbuild/build/install/include/yarp/dev/Drivers.h:10,
                 from /home/sdafarra/Software/robotology-superbuild/build/install/include/yarp/dev/PolyDriver.h:10,
                 from /home/sdafarra/Software/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:8:
/home/sdafarra/Software/robotology-superbuild/build/install/include/yarp/os/ConstString.h:434:12: note:   after user-defined conversion: yarp::os::ConstString::ConstString(const string&)
     inline ConstString(const std::string& str) {
            ^
In file included from /home/sdafarra/Software/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:10:0:
/home/sdafarra/Software/robotology-superbuild/build/install/include/yarp/dev/ControlBoardInterfaces.h:623:18: note:   initializing argument 2 of ‘virtual bool yarp::dev::IAxisInfo::getAxisName(int, yarp::os::ConstString&)’
     virtual bool getAxisName(int axis, yarp::os::ConstString& name) = 0;
                  ^
toolbox/CMakeFiles/WBToolbox.dir/build.make:302: recipe for target 'toolbox/CMakeFiles/WBToolbox.dir/src/base/RobotInterface.cpp.o' failed
make[5]: *** [toolbox/CMakeFiles/WBToolbox.dir/src/base/RobotInterface.cpp.o] Error 1
CMakeFiles/Makefile2:176: recipe for target 'toolbox/CMakeFiles/WBToolbox.dir/all' failed
make[4]: *** [toolbox/CMakeFiles/WBToolbox.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/WBToolbox.dir/build.make:111: recipe for target 'robotology/WBToolbox/CMakeFiles/YCMStamp/WBToolbox-build' failed
make[2]: *** [robotology/WBToolbox/CMakeFiles/YCMStamp/WBToolbox-build] Error 2
CMakeFiles/Makefile2:109: recipe for target 'CMakeFiles/WBToolbox.dir/all' failed
make[1]: *** [CMakeFiles/WBToolbox.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```
It has something to do with YARP wrapping STL strings.

traversaro, 2018-03-08:
What is the value of `YARP_WRAP_STL_STRING` on your YARP installation?

S-Dafarra, 2018-03-08:
If was ``ON``. Setting it to ``OFF`` solved the problem.

traversaro, 2018-03-08:
That's strange. On a new installation, the default value (at least for *nix OS) should be `OFF`.

S-Dafarra, 2018-03-08:
Probably I was having an old build.

diegoferigo, 2018-03-08:
The toolbox uses `std::string` and when it needs to interface with yarp methods that require `ConstString` everything works fine only when yarp maps it to `std::string`. This means `YARP_WRAP_STL_STRING=OFF`.

@S-Dafarra I was only partially right when we discussed f2f, [there's a way](https://github.com/robotology-playground/yarp-matlab-bindings/commit/1a215744e200f5e9a102e2fde363b0ebbca9e593) to check the status of this option which does not involve the exported cmake project.

Closing.

traversaro, 2018-03-08:
If that is the case, can you add a clear disclaimer such as https://github.com/robotology-playground/yarp-matlab-bindings/blob/master/CMakeLists.txt#L23 ?

Unfortunately `YARP_WRAP_STL_STRING` set to `OFF` is still the default for Windows users, and the compilation error would be quite critic for users to understand.

closed 2018-03-08 by traversaro

diegoferigo, 2018-03-08:
Yes I am planning to add exactly that in the coming `3.1` release.