robotology/wb-toolbox #99: Enhance the documentation on the website

diegoferigo, 2018-03-08:
Here's a list of possible topics:

- [ ] How to create a block (updated to `v3.1`)
- [ ] Automatic code generation pipeline
- [ ] How to solve errors on missing symbols
- [ ] How to update the library
- [ ] Example part 1: Develop a simple model that moves `iCub` joints in Gazebo
- [ ] Example part 2: Generate and run standalone code

diegoferigo, 2019-01-17:
The upcoming version of WBT should contain only tutorials on how to use the blocks since their creation is demanded by BlockFactory. All the tutorials that involve blocks creation and simulink libraries have been added / moved to BlockFactory.

traversaro, 2019-01-17:
Slightly OT, but (in addition to the project documentation) it would be good to link vvv-school tutorials that use WB-Toolbox, such as: 
* https://github.com/vvv-school/tutorial_impedance-control-simulink
* https://github.com/vvv-school/tutorial_wb-toolbox-gazebo-pendulum

diegoferigo, 2019-01-17:
It would be nice having a set of new tutorials here in WBT that address more a new user perspective. Here below some ideas:

1. **Basic Simulink simulation**: How to configure Simulink for a simple real-time compatible simulation. This could involve a _Discrete Filter_ block and a _Real-Time Synchronizer_, pointing out that we need to use a fixed step. This should provide a primer for new users.
1. **Use model blocks**: This tutorial should be focused on how to configure the environment e.g. with `YARP_DATA_DIRS` to find a `model.urdf`, and how to use the _Config_ block (in this case, no YARP-related blocks should be used).
1. **Simulink and Gazebo**: This tutorial should focus on Gazebo. It should handle all the YARP initialization, how to open Gazebo (with the plugin for synchronizing the clock), how to insert a model, few infos about the co-simulation approach we use, and how to send references and read measurement. It would be nice obtaining the same result of the vvv-school, that consisted in icub moving the elbow with a sine-wave reference.

This can be extremely beneficial with new members of the lab. @DanielePucci can we assign this task to someone?

@traversaro You anticipated my comment, I was writing the same thing while you commented ^^

DanielePucci, 2019-01-20:
@diegoferigo 
> This can be extremely beneficial with new members of the lab. @DanielePucci can we assign this task to someone?

As we discussed F2F, @aikolina may take upon this task

S-Dafarra, 2024-12-20:
This is quite outdated. Closing

closed 2024-12-20 by S-Dafarra