robotology/wb-toolbox #101: Model does not compile. Missing symbols

S-Dafarra, 2018-03-09:
I am trying to compile a model which contains only the ``Config`` block (configured via mask and the ``Real Time Synchronizer``, but Simulink is not able to compile, throwing the following error (after a long while)
```
Error while obtaining sizes from MEX S-function 'WBToolbox' in 'DumpJoints/Real Time Synchronizer'.
Caused by:
Invalid MEX-file '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64': Missing dependent shared libraries:
'libidyntree-modelio-urdf.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libidyntree-model.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libidyntree-core.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libidyntree-sensors.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libidyntree-model.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libidyntree-core.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-modelio-urdf.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libYARP_math.so.1' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libYARP_dev.so.1->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libYARP_sig.so.1' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libYARP_dev.so.1->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libYARP_OS.so.1' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libYARP_dev.so.1->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libYARP_sig.so.1' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libYARP_math.so.1->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libYARP_OS.so.1' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libYARP_math.so.1->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libYARP_OS.so.1' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libYARP_sig.so.1->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libidyntree-model.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-sensors.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libidyntree-core.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-sensors.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
'libidyntree-core.so' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-model.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'

Missing symbol '_ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEmmPKc' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEPKc' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEmmPKcm' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7reserveEm' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8_M_eraseEmm' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_appendEPKcm' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEED1Ev' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEE7_M_syncEPcmm' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEED1Ev' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEED1Ev' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEED1Ev' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZTTNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEE' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZTTNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEEE' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZTTNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEE' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZTVNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEEE' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZTVNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEE' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZTVNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEEE' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZTVNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEE' required by '/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZN8iDynTree11PositionRaw7inverseERKS0_' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZN8iDynTree11reportErrorEPKcS1_S1_' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZN8iDynTree12LinkAccArray6resizeERKNS_5ModelE' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZN8iDynTree12LinkAccArrayC1Ej' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
Missing symbol '_ZN8iDynTree12LinkAccArrayclEi' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
[... cut ...] ex/WBToolbox.mexa64'
Missing symbol '_ZN8iDynTree17PositionSemantics18setCoordinateFrameEi' required by '/home/sdafarra/Software/robotology-superbuild/build/install/lib/libidyntree-high-level.so->/home/sdafarra/Software/robotology-superbuild/build/install/mex/WBToolbox.mexa64'
[a lot more]
```

I have just installed from scratch the ``robotology_superbuild`` and run ``sudo ldconfig``. The matlab path and the .bashrc are set as in the [README](https://github.com/robotology/robotology-superbuild).

S-Dafarra, 2018-03-09:
Is the RPATH enabled on the toolbox?

traversaro, 2018-03-09:
All the models are failing to compile, or just the simple one? 
As a workaround, please append the directory containing the idyntree and yarp libraries to   `LD_LIBRARY_PATH`, and run `sudo ldconfig` .

S-Dafarra, 2018-03-09:
Right now I tried to add ``$ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX/lib`` to the ``LD_LIBRARY_PATH`` and build the superbuild from scratch. Once it finished I recompiled and reinstalled only the ``WBToolbox`` (making sure that it was actually recompiled) and run ``sudo ldconfig``. 
But when I try to compile the model I don't have anymore the error about the missing libraries, but I have a tons of error about missing symbols. I tried also to use other blocks, like the ``DOFsConverter`` but it happened as well.

S-Dafarra, 2018-03-09:
Interestingly I does not happen with only the ``Configuration`` block, but as soon I add another block. Additionally, if I open a blank new model, and I insert the ``Real Time Synchronizer`` I get the following error
[image]

traversaro, 2018-03-09:
This problem instead seems the usual https://github.com/robotology/codyco-superbuild/issues/141 .

S-Dafarra, 2018-03-09:
Thanks, I totally forgot about the reason why ``LD_PRELOAD=$LD_PRELOAD:/usr/lib/x86_64-linux-gnu/libstdc++.so.6`` was useful. I had two Matlab crashes in a row involving that library and I removed it. Now I am able to compile. I wonder whether it is necessary to add documentation about this :thinking:

S-Dafarra, 2018-03-09:
Also I would argue that the error message of https://github.com/robotology/codyco-superbuild/issues/141 was a bit more understandable :grin:

diegoferigo, 2018-03-10:
@S-Dafarra I opened https://github.com/robotology/wb-toolbox/issues/99 just two days ago :joy: It never occurred to me The Root Of Many Headache error (I used to call that way) for the missing `libstdc++` preload. The error you mentioned is more common.

However, the standard error is already discussed [in the documentation](https://robotology.github.io/wb-toolbox/mkdocs/troubleshooting/#problems-finding-libraries-and-libstdc). As soon as I add https://github.com/robotology/wb-toolbox/issues/99 I will cross reference TROMH to `libstdc++`.

P.s. In the future, tag me if you need a quick reply from me. Luckily @traversaro is always present.

closed 2018-03-10 by diegoferigo

S-Dafarra, 2018-03-10:
Thanks! Actually the behavior was very strange. It was working at the beginning, then started crashing continuously. Anyway, what about RPATH? It seems to me that this may cause issues when using the toolbox with the robotology superbuild. Am I right?

diegoferigo, 2018-03-10:
RPATH is enabled for long time now, what kind of problems are you mentioning?

S-Dafarra, 2018-03-10:
Sorry I did not see it initially in the CMakeLists. Initially I thought that that was the cause of the first set of errors of the [first comment](https://github.com/robotology/wb-toolbox/issues/101#issue-303798152). It was complaining about missing libraries and I thought about a linking problem. Was it all related to TROMH?

diegoferigo, 2018-03-10:
Understanding and debugging the causes of THROM is a difficult topic. When symbols are actually missing its raw output being so verbose is totally useless (it shows all external symbols, including standard library). I do not exclude that also in this case, even if all symbols are there, its output is hard to understand.

traversaro, 2018-03-10:
We should document THROM also from in the robotology-superbuild matlab section, as that is the entry-point for matlab software developed by us for most lab members.