robotology/wb-toolbox #102: Problem when using VOCAB_CM_POSITION control mode

traversaro, 2018-03-09:
When using the `VOCAB_CM_POSITION` control mode of the wb-toolbox with an eth-based iCub robot, this is the messages printed by the `yarprobotinterface` : 
~~~
17931	5295,720494	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17932	5295,730500	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17933	5295,740575	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17934	5295,750650	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17935	5295,760777	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17936	5295,770886	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17937	5295,780985	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17938	5295,791083	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17939	5295,801183	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17940	5295,811281	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17941	5295,821341	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17942	5295,831400	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17943	5295,841472	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17944	5295,851544	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17945	5295,861611	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17946	5295,871681	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17947	5295,881743	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17948	5295,891813	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17949	5295,901876	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17950	5295,911950	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17951	5295,922014	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17952	5295,932088	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17953	5295,942163	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
17954	5295,952224	WARNING	Performance warning: You are using positionMove commands at high rate (< 80  ms). Probably position control mode is not the right control mode to use.
~~~

The problem is that the setpoint for the position should be sent only when it changes, not at every simulink control loop.

diegoferigo, 2018-03-10:
Good to know, thanks Silvio for reporting!

I am wondering about how this might affect performance in position mode. @S-Dafarra did you do any experiment in Position Mode with the new toolbox version?

S-Dafarra, 2018-03-10:
Nope, I am not using the toolbox to control the robot in position right now, but we can do an easy test if you need 😉

traversaro, 2018-03-10:
Note that `VOCAB_CM_POSITION` is different from `VOCAB_CM_POSITION_DIRECT` .

S-Dafarra, 2018-03-10:
That's true. Thus, is it a problem of the toolbox or of the controller then? Does it make sense to address this at toolbox level?

traversaro, 2018-03-10:
How can this be addressed at the controller level?
 Once you placed the setReference block in your controller, I do not know it there is an easy way to make sure that the `output` function of the block is called only when the input signal is changing. 
I fear this need to either be fixed at the C++ level, or support for the `VOCAB_CM_POSITION` removed.

diegoferigo, 2018-03-10:
The C++ support of this custom logic only for `VOCAB_CM_POSITION` is a trivial change, I will address it soon. Users should not care about these technicalities.

diegoferigo, 2018-04-13:
@traversaro Would it make sense to extend this logic also to the other control modes?

traversaro, 2018-04-13:
Short answer: no.
Long answer: coming soon.

diegoferigo, 2018-05-11:
Fixed in PR https://github.com/robotology/wb-toolbox/pull/110

closed 2018-05-11 by diegoferigo