robotology/wb-toolbox #108: Exploit std::optional

diegoferigo, 2018-05-02:
In the toolbox we widely use the following pattern for achieving an exceptions-free error handling:

```cpp
bool func(const int param, MyObj& output)
```

```cpp
MyObj obj;
int param = 42;

if (!func(param, obj)) {
    wbtError << "Error!";
    return false;
}

... // Use obj
```

And sometimes we return a default-constructed object when the creation fails, which is all but safe if the method that actually checks the validity of the returned object is not called (e.g. `wbt::Signal`).

I recently read [Refactoring with C++17 std::optional ](https://www.bfilipek.com/2018/04/refactoring-with-c17-stdoptional.html) and I found the coming C++17's `std::optional` very useful for our use cases. The code above can be simplified as follows:

```cpp
std::optional<MyObj> func(const int param)
```

```cpp
int param = 42;

if (!(auto obj = func(param))) {
    wbtError << "Error!";
    return false;
}

... // Use obj
```

Waiting our switch to the new standard, I'd like to include an header-only library such as [optional-lite](https://github.com/martinmoene/optional-lite/) and take advantage of the undergoing refactoring of the `Block` / `BlockInformation` interfaces.

Any comment? @traversaro

---

A very common snippet is the following:

```cpp
    Signal output = blockInfo->getOutputPortSignal(0);
    if (!output.isValid()) {
        wbtError << "Output signal not valid.";
        return false;
    }
```

traversaro, 2018-05-03:
I like the use of `std::optional`, but consider that if `MyObj` is big or as an expensive constructor, returning it by value could (as part of a `std::optional`) could have a performance hit w.r.t. to accessing it by reference (depending on the case, this may be mitigated by [Return Value Optimization](https://en.wikipedia.org/wiki/Copy_elision) ).

Regarding the use of `optional-lite` or similar libraries (including also Google's abseil) I think this libraries are not suitable to use in the public interface of a non-header-only library, because their ABI depends on the compilation options of the downstream compilation unit that is including the headers, creating **runtime errors** for example if the library using `nonstd::optional` (wb-toolbox) is compiled with C++14 and the library that uses wb-toolbox is compiled with C++17.

traversaro, 2019-01-17:
cc @S-Dafarra related to our discussion of yesterday.

traversaro, 2019-01-17:
Btw, should we move this issue to blockfactory @diegoferigo ?

diegoferigo, 2019-01-17:
I would just close it. As soon as we switch to C++17 we might evaluate the usage of optional and other new containers shipped with the standard. Another good choice in my opinion would be using `std::variant` or `std::any` for the `Signal` class, I really have nightmares when I see the `void*` pointer to the internal buffer.

closed 2019-01-17 by diegoferigo