robotology/wb-toolbox #111: Not seeing the toolbox on MacOS 10

DanielePucci, 2018-05-08:
I've installed the `master` branch of the toolbox on my mac 
[image]
but, once in Simulink, I do not see the library

DanielePucci, 2018-05-08:
CC
@nunoguedelha

diegoferigo, 2018-05-09:
How did you [configure your environment](https://robotology.github.io/wb-toolbox/mkdocs/install/#configuration)?

diegoferigo, 2018-05-09:
In MacOs, after setting the `MATLABPATH` environment variable as described in the website, the library was showing up in the Simulink Library window, but it was empty. No blocks were displayed. However, opening it using `WBToolboxLibrary` worked, so the file is not corrupted.

It seems an error of the export of the `WBToolbox_library.slx` from the developer version `2017b` to the minimum supported version `2014b`.

I usually export it from my Ubuntu machine and commit it to GitHub, but for unknown reasons this exported version doesn't work properly in MacOs. However, exporting the library directly from MacOs (always using `2017b`) and installing the exported library, works as expected.

At the current moment I have no clue about the reason of this difference, it might be related to some mismatch in Matlab / Simulink between the different OSs. This is however a nice workaround and it is worth to be documented in the _Troubleshooting_ section of the website.

cc @francesco-romano @traversaro

diegoferigo, 2018-08-09:
@DanielePucci Probably the issue was related to the wrong path in the `startup_wbitoolbox.m` file pointing to the installed share directory. I fixed it in https://github.com/robotology/wb-toolbox/pull/148/commits/b4428e9b8a5fdcd65d1dd0156bfd9dbfb22203cd, can you please try this new version?

You should:

- Pull the latest changes of the toolbox
- Compile the project
- Install the project
- Execute again the `startup_wbitoolbox.m` 

After this please double check that the `shareDir` folder in the `pathdef.m` file created in the `userpath` folder contains correct (absolute) paths.

traversaro, 2018-12-22:
@DanielePucci have you tested again with a recent version of the WB-Toolbox?

diegoferigo, 2019-01-03:
If I recall this was an issue related to the path of the images used inside the Simulink Library. For the time being I close the issue since it should be fixed, @DanielePucci please open it again if you encounter any malfunctioning.

closed 2019-01-03 by diegoferigo

DanielePucci, 2019-01-04:
CC
@lrapetti