robotology/wb-toolbox #112: WBToolbox not compiling in devel branch

DanielePucci, 2018-05-08:
I was trying to compile the WBToolbox in `devel` branch. To do so, I set 
`YCM_EP_DEVEL_MODE_WBToolbox = ON`
and then I run `make` on the `robotology-superbuild`.

This is the error I get
~~~
[  5%] Linking CXX shared library ../lib/libMxAnyType.dylib
Undefined symbols for architecture x86_64:
  "_mxArrayToString", referenced from:
      MxAnyType::asString(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in MxAnyType.cpp.o
  "_mxFree", referenced from:
      MxAnyType::asString(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in MxAnyType.cpp.o
  "_mxGetCell_730", referenced from:
      MxAnyType::asAnyCell(std::__1::vector<std::__1::shared_ptr<AnyType>, std::__1::allocator<std::__1::shared_ptr<AnyType> > >&) in MxAnyType.cpp.o
  "_mxGetClassID", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetDimensions_730", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetFieldByNumber_730", referenced from:
      MxAnyType::asAnyStruct(std::__1::unordered_map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::shared_ptr<AnyType>, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::shared_ptr<AnyType> > > >&) in MxAnyType.cpp.o
  "_mxGetFieldNameByNumber", referenced from:
      MxAnyType::asAnyStruct(std::__1::unordered_map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::shared_ptr<AnyType>, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::shared_ptr<AnyType> > > >&) in MxAnyType.cpp.o
  "_mxGetM", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetN", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetNumberOfDimensions_730", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetNumberOfElements", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
      MxAnyType::asAnyCell(std::__1::vector<std::__1::shared_ptr<AnyType>, std::__1::allocator<std::__1::shared_ptr<AnyType> > >&) in MxAnyType.cpp.o
  "_mxGetNumberOfFields", referenced from:
      MxAnyType::asAnyStruct(std::__1::unordered_map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::shared_ptr<AnyType>, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::shared_ptr<AnyType> > > >&) in MxAnyType.cpp.o
  "_mxGetPr", referenced from:
      MxAnyType::asVectorDouble(std::__1::vector<double, std::__1::allocator<double> >&) in MxAnyType.cpp.o
  "_mxGetScalar", referenced from:
      MxAnyType::asScalar(double&) in MxAnyType.cpp.o
      MxAnyType::asInt(int&) in MxAnyType.cpp.o
      MxAnyType::asUInt(unsigned int&) in MxAnyType.cpp.o
      MxAnyType::asInt32(int&) in MxAnyType.cpp.o
      MxAnyType::asDouble(double&) in MxAnyType.cpp.o
  "_mxIsComplex", referenced from:
      MxAnyType::asVectorDouble(std::__1::vector<double, std::__1::allocator<double> >&) in MxAnyType.cpp.o
  "_mxIsDouble", referenced from:
      MxAnyType::asVectorDouble(std::__1::vector<double, std::__1::allocator<double> >&) in MxAnyType.cpp.o
  "_mxIsLogicalScalar", referenced from:
      MxAnyType::asBool(bool&) in MxAnyType.cpp.o
  "_mxIsLogicalScalarTrue", referenced from:
      MxAnyType::asBool(bool&) in MxAnyType.cpp.o
  "_mxIsNumeric", referenced from:
      MxAnyType::asScalar(double&) in MxAnyType.cpp.o
      MxAnyType::asInt(int&) in MxAnyType.cpp.o
      MxAnyType::asUInt(unsigned int&) in MxAnyType.cpp.o
      MxAnyType::asInt32(int&) in MxAnyType.cpp.o
      MxAnyType::asDouble(double&) in MxAnyType.cpp.o
  "_mxIsScalar", referenced from:
      MxAnyType::asScalar(double&) in MxAnyType.cpp.o
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
      MxAnyType::asInt(int&) in MxAnyType.cpp.o
      MxAnyType::asUInt(unsigned int&) in MxAnyType.cpp.o
      MxAnyType::asInt32(int&) in MxAnyType.cpp.o
      MxAnyType::asDouble(double&) in MxAnyType.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/libMxAnyType.0.1.dylib] Error 1
make[4]: *** [MxAnyType/CMakeFiles/MxAnyType.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [robotology/WBToolbox/CMakeFiles/YCMStamp/WBToolbox-build] Error 2
make[1]: *** [CMakeFiles/WBToolbox.dir/all] Error 2
make: *** [all] Error 2
****
~~~

@diegoferigo any idea on this?

DanielePucci, 2018-05-08:
As suggested by @diegoferigo, I run
~~~
make VERBOSE=1 -j1
~~~
in the build of the `robotology-superbuild`. I then get the following error
~~~
[  2%] Linking CXX shared library ../lib/libMxAnyType.dylib
cd /Users/dpucci/src/robotology-superbuild/build/robotology/WBToolbox/MxAnyType && /usr/local/Cellar/cmake/3.11.1/bin/cmake -E cmake_link_script CMakeFiles/MxAnyType.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++ -O3 -DNDEBUG -dynamiclib -Wl,-headerpad_max_install_names -current_version 0.1.0 -o ../lib/libMxAnyType.0.1.dylib -install_name @rpath/libMxAnyType.0.1.dylib CMakeFiles/MxAnyType.dir/src/MxAnyType.cpp.o 
Undefined symbols for architecture x86_64:
  "_mxArrayToString", referenced from:
      MxAnyType::asString(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in MxAnyType.cpp.o
  "_mxFree", referenced from:
      MxAnyType::asString(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in MxAnyType.cpp.o
  "_mxGetCell_730", referenced from:
      MxAnyType::asAnyCell(std::__1::vector<std::__1::shared_ptr<AnyType>, std::__1::allocator<std::__1::shared_ptr<AnyType> > >&) in MxAnyType.cpp.o
  "_mxGetClassID", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetDimensions_730", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetFieldByNumber_730", referenced from:
      MxAnyType::asAnyStruct(std::__1::unordered_map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::shared_ptr<AnyType>, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::shared_ptr<AnyType> > > >&) in MxAnyType.cpp.o
  "_mxGetFieldNameByNumber", referenced from:
      MxAnyType::asAnyStruct(std::__1::unordered_map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::shared_ptr<AnyType>, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::shared_ptr<AnyType> > > >&) in MxAnyType.cpp.o
  "_mxGetM", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetN", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetNumberOfDimensions_730", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
  "_mxGetNumberOfElements", referenced from:
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
      MxAnyType::asAnyCell(std::__1::vector<std::__1::shared_ptr<AnyType>, std::__1::allocator<std::__1::shared_ptr<AnyType> > >&) in MxAnyType.cpp.o
  "_mxGetNumberOfFields", referenced from:
      MxAnyType::asAnyStruct(std::__1::unordered_map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::shared_ptr<AnyType>, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::shared_ptr<AnyType> > > >&) in MxAnyType.cpp.o
  "_mxGetPr", referenced from:
      MxAnyType::asVectorDouble(std::__1::vector<double, std::__1::allocator<double> >&) in MxAnyType.cpp.o
  "_mxGetScalar", referenced from:
      MxAnyType::asScalar(double&) in MxAnyType.cpp.o
      MxAnyType::asInt(int&) in MxAnyType.cpp.o
      MxAnyType::asUInt(unsigned int&) in MxAnyType.cpp.o
      MxAnyType::asInt32(int&) in MxAnyType.cpp.o
      MxAnyType::asDouble(double&) in MxAnyType.cpp.o
  "_mxIsComplex", referenced from:
      MxAnyType::asVectorDouble(std::__1::vector<double, std::__1::allocator<double> >&) in MxAnyType.cpp.o
  "_mxIsDouble", referenced from:
      MxAnyType::asVectorDouble(std::__1::vector<double, std::__1::allocator<double> >&) in MxAnyType.cpp.o
  "_mxIsLogicalScalar", referenced from:
      MxAnyType::asBool(bool&) in MxAnyType.cpp.o
  "_mxIsLogicalScalarTrue", referenced from:
      MxAnyType::asBool(bool&) in MxAnyType.cpp.o
  "_mxIsNumeric", referenced from:
      MxAnyType::asScalar(double&) in MxAnyType.cpp.o
      MxAnyType::asInt(int&) in MxAnyType.cpp.o
      MxAnyType::asUInt(unsigned int&) in MxAnyType.cpp.o
      MxAnyType::asInt32(int&) in MxAnyType.cpp.o
      MxAnyType::asDouble(double&) in MxAnyType.cpp.o
  "_mxIsScalar", referenced from:
      MxAnyType::asScalar(double&) in MxAnyType.cpp.o
      MxAnyType::MxAnyType(mxArray_tag const*, bool) in MxAnyType.cpp.o
      MxAnyType::asInt(int&) in MxAnyType.cpp.o
      MxAnyType::asUInt(unsigned int&) in MxAnyType.cpp.o
      MxAnyType::asInt32(int&) in MxAnyType.cpp.o
      MxAnyType::asDouble(double&) in MxAnyType.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/libMxAnyType.0.1.dylib] Error 1
make[4]: *** [MxAnyType/CMakeFiles/MxAnyType.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [robotology/WBToolbox/CMakeFiles/YCMStamp/WBToolbox-build] Error 2
make[1]: *** [CMakeFiles/WBToolbox.dir/all] Error 2
make: *** [all] Error 2
~~~

diegoferigo, 2018-05-08:
Looks like the target `MxAnyType` doesn't link against the Matlab libraries, and indeed there's no `target_link_libraries` in its CMake configuration. However, this causes problems only on MacOs :thinking: 

Adding:

```cmake
target_link_libraries(${VARS_PREFIX} ${Matlab_MX_LIBRARY})
```

Should solve it.

traversaro, 2018-05-08:
> Looks like the target MxAnyType doesn't link against the Matlab libraries, and indeed there's no target_link_libraries in its CMake configuration. However, this causes problems only on MacOs thinking

This is related to weak/strong symbol linking, or something like that. I remember that at some point I found an option to the make the linker on Linux behave as the linker on macOS, but I forgot on which project I did this.

traversaro, 2018-05-08:
I think the option to obtain the same behavior on platform using ELF as Linux was `--unresolved-symbols=report-all` or `--no-allow-shlib-undefined` .

traversaro, 2018-05-08:
Related issue: https://github.com/robotology/codyco-superbuild/issues/9 .

DanielePucci, 2018-05-08:
After 22cf28f0c5359f72967813255837205ecfe410de and 5af0145ccafb22511d026ab73df6d9a5704a9107 I was able to compile the `devel` branch from the `robotology-superbuild`. Yet, #111 remains

diegoferigo, 2018-05-09:
After discussing f2f, it could be possible to add one of those linker flags in the CMakeLists only for Linux. Since _GNU ld_ cannot be used in macOS, the following should be enough:

```cmake
if(LINUX)
    set(CMAKE_SHARED_LINKER_FLAGS "--unresolved-symbols=report-all")
endif()
```

References:

- https://cmake.org/cmake/help/v2.8.12/cmake.html#variable:CMAKE_SHARED_LINKER_FLAGS
- https://stackoverflow.com/questions/2356168/force-gcc-to-notify-about-undefined-references-in-shared-libraries

traversaro, 2018-05-09:
To cover us from strange corner case, it would make to also add a check on the linker actually use by checking the last part of the [`CMAKE_CXX_LINK_EXECUTABLE`](https://cmake.org/cmake/help/v3.11/variable/CMAKE_LANG_LINK_EXECUTABLE.html) variable.

diegoferigo, 2018-05-09:
Unfortunately that is not what we need, it is the rule on how to link executables, and it contains something like this:

```
<CMAKE_CXX_COMPILER>  <FLAGS> <CMAKE_CXX_LINK_FLAGS> <LINK_FLAGS> <OBJECTS>  -o <TARGET> <LINK_LIBRARIES>
```

I still didn't find the right what to detect if `ld` is the linker executable.

traversaro, 2018-05-09:
I see. Let's keep like that then.

diegoferigo, 2018-05-09:
It is not really documented, but [`CMAKE_LINKER`](https://cmake.org/pipermail/cmake/2014-August/058271.html) might work.

```cpp
if(UNIX AND NOT APPLE)
    get_filename_component(LINKER_BIN ${CMAKE_LINKER} NAME)
    if(${LINKER_BIN} EQUAL "ld")
        set(CMAKE_SHARED_LINKER_FLAGS "--unresolved-symbols=report-all")
    endif()
endif()
```

traversaro, 2018-07-26:
CMake issue related to identification of the used linker : https://gitlab.kitware.com/cmake/cmake/issues/16543#note_405513 .