robotology/wb-toolbox #113: Warning during linking

DanielePucci, 2018-05-08:
After #111 and #112, I tried to clone and compile the `WBToolbox` only.

I got the following warning after compilation

~~~
[100%] Linking CXX shared library WBToolbox.mexmaci64
ld: warning: direct access in function 'wbt::DiscreteFilter::initializeInitialConditions(wbt::BlockInformation const*)' from file 'CMakeFiles/WBToolbox.dir/src/DiscreteFilter.cpp.o' to global weak symbol 'typeinfo for iCub::ctrl::IFilter' from file '/Users/dpucci/src/robotology-superbuild/build/install/lib/libctrlLib.a(filters.cpp.o)' means the weak symbol cannot be overridden at runtime. This was likely caused by different translation units being compiled with different visibility settings.
~~~

diegoferigo, 2018-05-08:
`IFilter` is an abstract class which is implemented by all the filters in `iCub::ctrl`. If I list the exported symbols of the library, in my Ubuntu system I have:

```
dferigo@8133d34144f0:~> nm -C /iit/local/lib/libctrlLib.a | grep IFil
0000000000000000 W iCub::ctrl::IFilter::IFilter()
0000000000000000 W iCub::ctrl::IFilter::IFilter()
0000000000000000 n iCub::ctrl::IFilter::IFilter()
0000000000000000 W iCub::ctrl::IFilter::~IFilter()
0000000000000000 W iCub::ctrl::IFilter::~IFilter()
0000000000000000 W iCub::ctrl::IFilter::~IFilter()
0000000000000000 n iCub::ctrl::IFilter::~IFilter()
0000000000000000 V typeinfo for iCub::ctrl::IFilter
0000000000000000 V typeinfo name for iCub::ctrl::IFilter
0000000000000000 V vtable for iCub::ctrl::IFilter
```

This warning looks related to the visibility of such symbols, and it might be solved setting the `-fvisibility` linker flag.

@DanielePucci Can you try to manually add the `-fvisibility=hidden` flag?
@traversaro any insights on this due to your experience?

Resources:
- https://developer.apple.com/library/content/documentation/DeveloperTools/Conceptual/CppRuntimeEnv/Articles/SymbolVisibility.html
- https://stackoverflow.com/questions/9894961/strange-warnings-from-the-linker-ld

diegoferigo, 2018-05-08:
Not strictly. This issue is a warning about `icub-main` and `wb-toolbox` symbol visibility, and hence maybe related to the compiler flags, the other one a linking error due to a forgotten `target_link_libraries`. The reason why on Linux the latter was working is likely a difference in the compiler options as you described in #112.

diegoferigo, 2019-01-17:
Closing this stale issue. @DanielePucci let me know if this warning shows up again.

closed 2019-01-17 by diegoferigo