robotology/wb-toolbox #118: Provide a block for setting/getting motor control parameters(K_tau, K_bemf)

vibhoraggarwal, 2018-05-23:
**Desription**:
- Currently, we use `yarpmotorgui`, to set the motor control parameters like $K_\tau$ and $K_{bemf}$. It would be beneficial for many users if we can set/get the motor control parameters from the Simulink.

vibhoraggarwal, 2018-05-23:
cc: @diegoferigo

diegoferigo, 2018-05-28:
@vibhoraggarwal I opened PR https://github.com/robotology/wb-toolbox/pull/124 with the new options. Please test it and let me know if it works as expected. I couldn't test it on gazebo since the motor interfaces are not implemented and they return `false`.

vibhoraggarwal, 2018-05-28:
cc: @diegoferigo 
Update:
1. `Get Motor Measurement` block works **fine**
2. There is an **issue** running `set motor references` block:
     - block looks for a variable called: `refSpeed`, which is not defined, giving the following error
     - block fails to set the `control mode` and gives following error

```
Error 1:
Invalid setting in 'SquareTorque2016b/DataCollectin/T/Subsystem1/Set Motor References1/S-Function' for parameter 'refSpeed'.
     Caused by:
     Error evaluating parameter 'refSpeed' in 'SquareTorque2016b/DataCollectin/T/Subsystem1/Set Motor   References1/S-Function'
    Undefined function or variable 'refSpeed'.```

Error 2:
An error occurred while running the simulation and the simulation was terminated
Caused by:
Error reported by S-function 'WBToolbox' in 'SquareTorque2016b/DataCollectin/T/Subsystem1/Set Motor References1/S-Function':
Failed to set control mode.

vibhoraggarwal, 2018-05-28:
Yes, `set motor references` block work well now. 

The problem related to setting the `control mode` in the [above comment](https://github.com/robotology/wb-toolbox/issues/118#issuecomment-392527141) is not related to the block, but toolbox in general
cc: @diegoferigo

vibhoraggarwal, 2018-05-28:
The recent commit: https://github.com/diegoferigo/WB-Toolbox/commit/538d55eebe6f32f46b646cc9ac4535200992dd52
did not solve the issue of setting the `control mode` but. was crashing the Matlab

diegoferigo, 2018-05-29:
@vibhoraggarwal That commit was still a draft. I caught an hidden bug and fixed it in #126. The branch with the new block has been rebased on top of it and it is ready to be tested again.

vibhoraggarwal, 2018-05-30:
I also have this message in the `yarplogger`:

```15110	2999,244711	ERROR	positionMoveRaw: skipping command because  BOARD left_leg-eb6-j0_3 (IP 10.0.1.6)   joint  0  is not in VOCAB_CM_POSITION mode```

diegoferigo, 2018-05-30:
@vibhoraggarwal I rebased this branch of top of the fix for the error you reported.

vibhoraggarwal, 2018-05-30:
It still fails to set references using the position control mode
```An error occurred while running the simulation and the simulation was terminated
Caused by:
Error reported by S-function 'WBToolbox' in 'testModel/Set References/S-Function':
Failed to set references.```

vibhoraggarwal, 2018-05-30:
1. The `setReference` block for setting the position reference works fine. But,
2. `SetMotorReference` for setting lwo-level motor constants does not work, and gives the following error:

```An error occurred while running the simulation and the simulation was terminated
Caused by:
Error reported by S-function 'WBToolbox' in 'SquareTorque2016b/DataCollectin/T/Subsystem1/Set Motor References1/S-Function':
Failed to set control mode.

vibhoraggarwal, 2018-05-31:
`Set Motor References` in case of setting of $K_{bemf}$ is not working fine. The block is unable to overwrite  $K_{bemf}$'s value.
cc: @diegoferigo

vibhoraggarwal, 2018-06-15:
@diegoferigo as talked f2f, the `yarpmotorgui` does show me updated values of the parameters. 
But there is a issue while sending the $K_\tau$ and $K_{bemf}$

diegoferigo, 2018-06-20:
@vibhoraggarwal Please try again https://github.com/diegoferigo/WB-Toolbox/tree/feature/motorParametersBlock on the robot. I expect now that at least `Ktau` to work. The code is similar to the previous one, this means that `Bemf` might not work yet.

vibhoraggarwal, 2018-06-20:
@diegoferigo I checked `Set Motor Parameters` block. It works fine for setting up of all the three parameters.

diegoferigo, 2018-06-20:
I'm glad to hear it! The reason why before Kbemf was not working will remain unsolved, I don't mind :smile:

Thanks for testing. I close this issue as soon as I open and merge a new PR with the branch you tested.

diegoferigo, 2018-07-04:
Implemented in PR https://github.com/robotology/wb-toolbox/pull/135.

PR #135 (2018-06-25): New Set Motor Parameters block
This block substitutes _SetLowLevelPID_ and introduces the following changes:

- `P` `I` `D` are now input signals. They must have a width equal to the controlled number of joints.
- Now is possible to set from the mask the motor constant `Ktau` and the motor back EMF constant.
- Removed the previous `PID` and `PIDConfiguration` matlab classes.

This new block allows doing gain scheduling during runtime :tada: The new PID values are applied only if the new values differ from the previous, avoiding many slow rpc calls.

Closes #118.

cc @DanielePucci @traversaro @ [... cut]
closed 2018-07-04 by diegoferigo