robotology/wb-toolbox #119: Error compiling due to deprecation of control modes in YARP

hu-yue, 2018-05-23:
After updating YARP to the latest version, which deprecated the control modes ending with `2`, I got errors when compiling the toolbox, as below:

```
In file included from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/ControlBoardInterfaces.h:22:0,
                 from /usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:10:
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/IControlMode2.h:19:81: error: conflicting declaration ‘typedef class yarp::dev::IControlMode yarp::dev::IControlMode2’
 YARP_DEPRECATED_TYPEDEF_MSG("Use yarp::dev::IControlMode instead") IControlMode IControlMode2;
                                                                                 ^
In file included from /usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:1:0:
/usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/include/base/RobotInterface.h:20:15: note: previous declaration as ‘class yarp::dev::IControlMode2’
         class IControlMode2;
               ^
In file included from /usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:11:0:
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/IControlLimits2.h:17:85: error: conflicting declaration ‘typedef class yarp::dev::IControlLimits yarp::dev::IControlLimits2’
 YARP_DEPRECATED_TYPEDEF_MSG("Use yarp::dev::IControlLimits instead") IControlLimits IControlLimits2;
                                                                                     ^
In file included from /usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:1:0:
/usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/include/base/RobotInterface.h:23:15: note: previous declaration as ‘class yarp::dev::IControlLimits2’
         class IControlLimits2;
               ^
[ 23%] Building CXX object toolbox/CMakeFiles/WBToolbox.dir/src/YarpRead.cpp.o
In file included from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h:15:0,
                 from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/PolyDriver.h:13,
                 from /usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:8:
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h: In instantiation of ‘bool yarp::dev::DeviceDriver::view(T*&) [with T = yarp::dev::IControlMode2]’:
/usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:546:13:   required from ‘T* wbt::RobotInterface::getInterfaceFromTemplate(T*&) [with T = yarp::dev::IControlMode2]’
/usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:271:72:   required from here
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:84:53: error: cannot dynamic_cast ‘yarp::dev::DeviceDriver::getImplementation()’ (of type ‘class yarp::dev::DeviceDriver*’) to type ‘class yarp::dev::IControlMode2*’ (target is not pointer or reference to complete type)
         T *v = dynamic_cast<T *>(getImplementation());
                                                     ^
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h: In instantiation of ‘bool yarp::dev::DeviceDriver::view(T*&) [with T = yarp::dev::IControlLimits2]’:
/usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:546:13:   required from ‘T* wbt::RobotInterface::getInterfaceFromTemplate(T*&) [with T = yarp::dev::IControlLimits2]’
/usr/local/src/robot/robotology-superbuild/robotology/WBToolbox/toolbox/src/base/RobotInterface.cpp:327:74:   required from here
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/DeviceDriver.h:84:53: error: cannot dynamic_cast ‘yarp::dev::DeviceDriver::getImplementation()’ (of type ‘class yarp::dev::DeviceDriver*’) to type ‘class yarp::dev::IControlLimits2*’ (target is not pointer or reference to complete type)
toolbox/CMakeFiles/WBToolbox.dir/build.make:302: recipe for target 'toolbox/CMakeFiles/WBToolbox.dir/src/base/RobotInterface.cpp.o' failed
make[5]: *** [toolbox/CMakeFiles/WBToolbox.dir/src/base/RobotInterface.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
```

Thanks @diegoferigo

traversaro, 2018-05-23:
I think substituting the forward declaration in https://github.com/robotology/wb-toolbox/blob/932c81cd049c7ce0c2494d60fe52bfd8c11dfd20/toolbox/include/base/RobotInterface.h#L23 with a proper include everything should work fine both with YARP master and YARP devel .

traversaro, 2018-05-24:
cc @S-Dafarra @gabrielenava

diegoferigo, 2018-05-25:
Fixed in #120.

closed 2018-05-25 by diegoferigo