robotology/wb-toolbox #127: Matlab crash in Ubuntu 16.04

lrapetti, 2018-05-29:
I have installed Matlab R2018a in Ubuntu 16.04, but it keep crashing reporting:
[image]
```

MATLAB crash file:/home/lrapetti/matlab_crash_dump.14993-1:

--------------------------------------------------------------------------------
       Segmentation violation detected at mar mag 29 11:22:20 2018 +0200
--------------------------------------------------------------------------------

Configuration:
  Crash Decoding           : Disabled - No sandbox or build area path
  Crash Mode               : continue (default)
  Default Encoding         : UTF-8
  Deployed                 : false
  Desktop Environment      : Unity
  GNU C Library            : 2.23 stable
  Graphics Driver          : Unknown hardware 
  Java Version             : Java 1.8.0_144-b01 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  MATLAB Architecture      : glnxa64
  MATLAB Entitlement ID    : 5113464
  MATLAB Root              : /usr/local/MATLAB/R2018a
  MATLAB Version           : 9.4.0.813654 (R2018a)
  OpenGL                   : hardware
  Operating System         : Ubuntu 16.04.4 LTS
  Process ID               : 14993
  Processor ID             : x86 Family 6 Model 61 Stepping 4, GenuineIntel
  Session Key              : 700e5607-7606-451a-b52c-695fbe733630
  Static TLS mitigation    : Enabled: Full
  Window System            : The X.Org Foundation (11905000), display :0

Fault Count: 1

Abnormal termination

Register State (from fault):
  RAX = 0000000000000000  RBX = 00007f03e40ab100
  RCX = 0000000000000000  RDX = 0000000000000000
  RSP = 00007f038d47c1a0  RBP = 00007f03e40a9650
  RSI = 00007f03e4000048  RDI = 0000000000000001

   R8 = 00007f03e40ab130   R9 = 0000000000000000
  R10 = 00007f03e44bd570  R11 = 00007f048db864f0
  R12 = 00007f03e44bd430  R13 = 00000000000009a0
  R14 = 000000000000002c  R15 = 00007f03e44bd520

  RIP = 00007f048db86512  EFL = 0000000000010202

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007f048db86512                    /lib/x86_64-linux-gnu/libc.so.6+00541970 cfree+00000034
[  1] 0x00007f04802a3254                     /usr/lib/nvidia-384/libGL.so.1+00549460

```
The crash happen at different moments: sometimes during the startup of matlab, and when simulink models are launched. Pressing the button `Attempt to continue` I am still able to work even after it crashes.

Even if this problem may not be related to the use of `wb-toolbox`, the solution may be relevant for those using it.

lrapetti, 2018-05-29:
updated previous comment with the error report.

traversaro, 2018-05-29:
What is the value of the `YARP_DATA_DIRS` and `YARP_ROBOT_NAME` environmental variables? See http://www.yarp.it/yarp_env_variables.html for the meaning of this variables. 

~~~
yarp resource --find model.urdf
~~~
Is able to find the model?

lrapetti, 2018-05-29:
```
$YARP_DATA_DIRS
:/home/lrapetti/robotology-superbuild/build/install/share/yarp:
/home/lrapetti/robotology-superbuild/build/install/share/iCub:
/home/lrapetti/element_torque-control-via-current/software/jointTorqueControl/build/install/share/yarp:
/home/lrapetti/element_torque-control-via-current/software/motorCurrentControl/build/install/share/yarp
```

`YARP_ROBOT_NAME` instead is empty, and the model is not found by `yarp resource --find model.urdf`

traversaro, 2018-05-29:
I think you should just set `YARP_ROBOT_NAME` to the robot that you are controlling, in the case of iCub in gazebo installed by https://github.com/robotology-playground/icub-models it should be `iCubGazeboV2_5`, while for the model installed by https://github.com/robotology/icub-gazebo it is `icubGazeboSim` .

lrapetti, 2018-05-29:
* The issue with libGL was solved by:
> - Terminal **sudo matlab**
> - Command Window **opengl('save','software')** 

* The issue in loading the .urdf was solved by:
> I think you should just set YARP_ROBOT_NAME to the robot that you are controlling, in the case of iCub in gazebo installed by https://github.com/robotology-playground/icub-models it should be [...]

diegoferigo, 2018-05-29:
I reopen the issue because the toolbox shouldn't crash but warn the user that the model is not found. Related to #69.

lrapetti, 2018-05-30:
> The issue in loading the .urdf was solved by:
>
>     I think you should just set YARP_ROBOT_NAME to the robot that you are controlling, in the case of iCub in gazebo installed by [...]

The solution found here was not effective since the problem is still present even when the YARP_ROBOT_NAME is set and the .urdf is found through `yarp resource --from model.urdf`

traversaro, 2018-05-30:
Are you sure that `YARP_ROBOT_NAME` set in the `.bashrc` is found in matlab? You should be able to get it using the `getenv` matlab function if this is the case.

lrapetti, 2018-05-30:
> Are you sure that YARP_ROBOT_NAME set in the .bashrc is found in matlab? You should be able to get it using the getenv matlab function if this is the case.

This is what I get:
```

getenv('YARP_ROBOT_NAME')

ans =

    'icubGazeboSim'

```

lrapetti, 2018-05-31:
@traversaro @diegoferigo 
[SOLVED]
I have test the model after installing the robotology-playground [icub-models](https://github.com/robotology-playground/icub-models) and it worked correctly. So I have figured out that the problem was that the simulink model was internally setting the `YARP_ROBOT_NAME` to `iCubGazeboV_5`.
This is proved by checking the YARP_ROBOT_NAME after running the simulation:

```
getenv('YARP_ROBOT_NAME')

ans =

    'iCubGazeboV_5'

```

My specific issue is solved, but the issue remain open related to #69

diegoferigo, 2018-06-12:
I couldn't manage to reproduce this issue:

[image]

I did unset the `YARP_ROBOT_NAME` from the terminal from which I executed matlab, but the crash didn't happen and the error I would expect to read showed up.

lrapetti, 2018-06-12:
> I did unset the YARP_ROBOT_NAME from the terminal from which I executed matlab, but the crash didn't happen and the error I would expect to read showed up.
> 
> This happens even if I set YARP_ROBOT_NAME=WrongRobotName.

@diegoferigo 
Same for me, I am no longer experiencing the crash and I get the error message:
`ResourceFinder couldn't find urdf file model.urdf.`

Right now I am attached at https://github.com/diegoferigo/WB-Toolbox/tree/feature/motorConstants

diegoferigo, 2018-06-12:
Awesome, you were on the master branch when you opened this issue, right? Probably the singleton refactoring of PR #115 solved it since it changed exactly the piece of code that finds the model and creates a `KinDynComputations` object from that.

closed 2018-06-12 by diegoferigo