robotology/wb-toolbox #128: Set References 'Position' and 'Current' not working

lrapetti, 2018-06-01:
### Model
I'm trying to run the simple following simulink model:
[image]

using icubGazeboSim with the `config`:
```
% Robot configuration for WBT3.0
WBTConfigRobot           = WBToolbox.Configuration;
WBTConfigRobot.RobotName = 'icubSim';
WBTConfigRobot.UrdfFile  = 'model.urdf';
WBTConfigRobot.LocalName = 'WBT';

WBTConfigRobot.ControlBoardsNames = {'torso','left_arm','right_arm','left_leg','right_leg'};
WBTConfigRobot.ControlBoardsNames = {'left_leg','right_leg'};
WBTConfigRobot.ControlledJoints   = {'l_hip_pitch','l_hip_roll','l_hip_yaw','l_knee','l_ankle_pitch','l_ankle_roll', ...
                                     'r_hip_pitch','r_hip_roll','r_hip_yaw','r_knee','r_ankle_pitch','r_ankle_roll'};
```

### Error
It works with all the control configuration a part from `Position` and `Current` that return the errors:
* `Current`
```
An error occurred while running the simulation and the simulation was terminated
Caused by:
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/Set References/S-Function':

[mdlOutputs]
Failed to set control mode.
Component:Simulink | Category:Block error
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/Set References/S-Function':

[mdlTerminate]
Failed to set control mode.
Failed to release the RemoteControlBoardRemapper.
```
* `Position`
```
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/Set References/S-Function':

[mdlStart]
Failed to initialize speed references.
Component:Simulink | Category:Block error
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/synchronizer/GAZEBO_SYNCHRONIZER/Simulator Synchronizer':

[mdlTerminate]
Component:Simulink | Category:Block error
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/synchronizer/REAL_TIME_SYNC/Real Time Synchronizer':

[mdlTerminate]
Component:Simulink | Category:Block error
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/Set References/S-Function':

[mdlTerminate]
Failed to set control mode.
Failed to release the RemoteControlBoardRemapper.
Component:Simulink | Category:Block error
```

diegoferigo, 2018-06-01:
For what concerns current control, the master branch provided just a partial support. I remember I added those control modes but they never got tested on iCub. The `devel` branch is providing full support to motor-side quantities, and it will be soon merged into master (right after Yarp 3.0).

I have to dig more on the _Position_ mode, I remember @vibhoraggarwal used it not long ago.

lrapetti, 2018-06-02:
> For what concerns current control, the master branch provided just a partial support. I remember I added those control modes but they never got tested on iCub. The devel branch is providing full support to motor-side quantities, and it will be soon merged into master (right after Yarp 3.0).

I have seen at https://github.com/robotology/wb-toolbox/pull/110 the implementation for the block at the motor side. I have seen changes in getting the measurement but not in the setting of current from the master https://github.com/robotology/wb-toolbox/blob/master/toolbox/src/SetReferences.cpp#L304. 
Anyway I will test the `devel` and let you know if it prevents getting the error.

lrapetti, 2018-06-04:
Now I am using the `devel` branch, but I am still getting errors:
* using `Set References` in `Position` control:
```
An error occurred while running the simulation and the simulation was terminated
Caused by:
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/Set References/S-Function':
Failed to set references.
Component:Simulink | Category:Block error
```
* using the `Set Motor References`
```
Invalid setting in 'SimulinkMotorCurrentControl/Set Motor References/S-Function' for parameter 'refSpeed'.
Caused by:
Error evaluating parameter 'refSpeed' in 'SimulinkMotorCurrentControl/Set Motor References/S-Function'
Undefined function or variable 'refSpeed'.
Variable 'refSpeed' has been deleted from base workspace.
```

Those two problems are discussed in https://github.com/robotology/wb-toolbox/issues/118 and a solution has been proposed.
However, even in the `refSpeed` error is solved, setting the `Set Motor References` in `Current` control causes Gazebo to terminate and raises the error:
```
An error occurred while running the simulation and the simulation was terminated
Caused by:
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/synchronizer/GAZEBO_SYNCHRONIZER/Simulator Synchronizer':
Error connecting to simulator clock server.
Component:Simulink | Category:Block error
Error reported by S-function 'WBToolbox' in 'SimulinkMotorCurrentControl/Set Motor References/S-Function':
n� >
```

diegoferigo, 2018-06-05:
> using `Set References` in `Position control`

I should check your setup for what concern the first issue. I just tried on a similar controller and I can use `SetReferences` in `Position` mode. There was an issue some time ago fixed in https://github.com/robotology/wb-toolbox/commit/9ed80812f7b5b77fecbbf47493a05c5712600d08.

> using the `Set Motor References`

I checked upstream and `ICurrentControl` is not implemented, hence it fails. I know you are working on this implementation, I can provide support to debug the problem. As discussed f2f, if gazebo crashes the problem is likely related to gazebo-yarp-plugins rather than wb-toolbox.

traversaro, 2018-06-05:
> I checked upstream and ICurrentControl is not implemented, hence it fails. 

Is not implemented in which class/device?

lrapetti, 2018-06-05:
> As discussed f2f, if gazebo crashes the problem is likely related to gazebo-yarp-plugins rather than wb-toolbox

It may be related to gazebo-yarp-plugins, but it works fine using the `yarpmotorgui`.
This is the Gazebo output before it get's closed when using `Current`, compared to the output of the `PWM` control (that instead is working fine):
* Current
```
[DEBUG]Initializing Trajectory Generator with current values 
[INFO]created device <gazebo_controlboard>. See C++ class GazeboYarpControlBoardDriver for documentation.
yarp: Port /icubSim/applyExternalWrench/rpc:i active at tcp://10.255.111.97:10041/
yarp: Receiving input from /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/left_leg/rpc:o to /icubSim/left_leg/rpc:i using tcp
yarp: Receiving input from /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/left_leg/command:o to /icubSim/left_leg/command:i using udp
yarp: Sending output from /icubSim/left_leg/stateExt:o to /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/left_leg/stateExt:i using udp
yarp: Receiving input from /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/right_leg/rpc:o to /icubSim/right_leg/rpc:i using tcp
yarp: Receiving input from /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/right_leg/command:o to /icubSim/right_leg/command:i using udp
yarp: Sending output from /icubSim/right_leg/stateExt:o to /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/right_leg/stateExt:i using udp
```

* PWM
```
[DEBUG]Initializing Trajectory Generator with current values 
[INFO]created device <gazebo_controlboard>. See C++ class GazeboYarpControlBoardDriver for documentation.
yarp: Port /icubSim/applyExternalWrench/rpc:i active at tcp://10.255.111.97:10041/
yarp: Receiving input from /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/left_leg/rpc:o to /icubSim/left_leg/rpc:i using tcp
yarp: Receiving input from /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/left_leg/command:o to /icubSim/left_leg/command:i using udp
yarp: Sending output from /icubSim/left_leg/stateExt:o to /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/left_leg/stateExt:i using udp
yarp: Receiving input from /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/right_leg/rpc:o to /icubSim/right_leg/rpc:i using tcp
yarp: Receiving input from /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/right_leg/command:o to /icubSim/right_leg/command:i using udp
yarp: Sending output from /icubSim/right_leg/stateExt:o to /WBT/SimulinkMotorCurrentControlConfiguration/icubSim/right_leg/stateExt:i using udp
yarp: Receiving input from /WBT_synchronizer/rpc:o to /clock/rpc using tcp
```

@diegoferigo Is it possible that the problem is related to `WB-toolbox syncronizer`?

lrapetti, 2018-06-06:
> * wb-toolbox uses `setRefCurrents(const double* currs)` method, while yarpmotorgui uses `setRefCurrent(int m, double curr)`

I just tested `setRefCurrents(const double* currs)` with _gazebo-yarp-pluging_ and it worked fine when not in _wb-toolbox_.

traversaro, 2018-06-06:
In the `controlboardremapper` device, the `setRefCurrents(const double* currs)` seems to be correctly implement, see https://github.com/robotology/yarp/blob/devel/src/libYARP_dev/src/devices/ControlBoardRemapper/ControlBoardRemapper.cpp#L4409  .

lrapetti, 2018-06-07:
I have verified that the call of `SetRefCurrents(const double* currs)` done in the [ControlBoardRemapepr](https://github.com/robotology/yarp/blob/devel/src/libYARP_dev/src/devices/ControlBoardRemapper/ControlBoardRemapper.cpp#L4446) is executed and returns true, but on the other side there is no call of `setRefCurrents(const int n_joint, const int *joints, const double *t)` in the [GazeboYarpControlBoard](https://github.com/robotology/gazebo-yarp-plugins/blob/devel/plugins/controlboard/src/ControlBoardDriverCurrent.cpp#L52).

### Solution
The problem can be bypassed by implementing in the ControlBoardRemapper `SetRefCurrents(const double* currs)` the call to the method `setRefCurrent(int j, double v)` such as it is done in [`setRefDutyCycles(const double* refs)`](https://github.com/robotology/yarp/blob/devel/src/libYARP_dev/src/devices/ControlBoardRemapper/ControlBoardRemapper.cpp#L4174):
```
bool ControlBoardRemapper::setRefCurrents(const double* currs)
{
    return ret;

    bool ret=true;

    for(int l=0;l<controlledJoints;l++)
    {
        int off=(int)remappedControlBoards.lut[l].axisIndexInSubControlBoard;
        size_t subIndex=remappedControlBoards.lut[l].subControlBoardIndex;

        yarp::dev::RemappedSubControlBoard *p=remappedControlBoards.getSubControlBoard(subIndex);

        if (!p)
        {
            return false;
        }

        if (p->iPwm)
        {
            bool ok = p->iCurr->setRefCurrent(off, currs[l]);
            ret = ret && ok;
        }
        else
        {
            ret=false;
        }
    }

    return ret;

```

### Open Issue
It still has to be understood why the `ControlBoardRemapper` is not colling correctly the `setRefCurrents(const int n_joint, const int *joints, const double *t)` method.
I found that [`setRefCurrents(const int n_joint, const int *joints, const double *t)`](https://github.com/robotology/gazebo-yarp-plugins/blob/devel/plugins/controlboard/src/ControlBoardDriverCurrent.cpp#L52) is not implemented correctly in the `GazeboYarpControlBoard`.
[`ret = setRefCurrent(joints[i], t[i]);`](https://github.com/robotology/gazebo-yarp-plugins/blob/devel/plugins/controlboard/src/ControlBoardDriverCurrent.cpp#L57)
should be replaced by:
`m_jntReferenceTorques[joints[i]] = t[i] * m_kPWM[joints[i]];`
but still not resolving the error

Should we move the whole issue to https://github.com/robotology/gazebo-yarp-plugins or https://github.com/robotology/yarp?

traversaro, 2018-06-07:
Great insight @lrapetti ! Let's discuss f2f after lunch how to actually find the real problem.

diegoferigo, 2018-06-07:
@traversaro I had in mind the [iCub implementation](https://github.com/robotology/icub-main/blob/4f5881acafab693f763dd277f87a8996fdea01a3/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L4730-L4753). Here we are talking about gazebo-yarp-plugins, sorry for the confusion.

@lrapetti I'm glad you found the culprit! You can open a PR directly in gazebo-yarp-plugins when your fix is ready, and link this issue from there. No need to open new issues also in the other repository.

lrapetti, 2018-06-07:
The problem is fixed at https://github.com/robotology/yarp/pull/1734.
The message sent with `setRefCurrents(const int n_joint, const int *joints, const double *t)` in the `ControlBoardRemapper` was not parsed correctly.

closed 2018-06-07 by lrapetti

diegoferigo, 2018-06-12:
Awesome, well done @lrapetti :tada: