robotology/wb-toolbox #129: Matlab 2018a compilation problem using the devel branch

lrapetti, 2018-06-04:
~~~
lrapetti@iiticublap048:~/robotology-superbuild/build/robotology/WBToolbox$ make
-- Found YCM: /usr/local/share/YCM (found version "0.6.0.16-20180507.2+git4dbe172")
-- Found Matlab: /usr/local/MATLAB/R2018a/extern/include (found version "unknown") found components:  MX_LIBRARY ENG_LIBRARY MAIN_PROGRAM 
-- Found Matlab: /usr/local/MATLAB/R2018a/extern/include (found version "unknown") found components:  MX_LIBRARY 
-- Using iCub from install
-- Setting up installation of codyco.ini to /home/lrapetti/robotology-superbuild/build/install/share/yarp/config/path.d folder.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lrapetti/robotology-superbuild/build/robotology/WBToolbox
[  4%] Built target MxAnyType
[  9%] Built target ClockServer
[ 85%] Built target WBToolboxLibrary
[ 90%] Built target WBToolboxCoder
[ 92%] Linking CXX shared library ../lib/WBToolbox.mexa64
/usr/local/MATLAB/R2018a/bin/glnxa64/libCppMicroServices.so.3.1.1: undefined reference to `std::thread::_State::~_State()@GLIBCXX_3.4.22'
/usr/local/MATLAB/R2018a/bin/glnxa64/libCppMicroServices.so.3.1.1: undefined reference to `std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)())@GLIBCXX_3.4.22'
/usr/local/MATLAB/R2018a/bin/glnxa64/libCppMicroServices.so.3.1.1: undefined reference to `typeinfo for std::thread::_State@GLIBCXX_3.4.22'
collect2: error: ld returned 1 exit status
toolbox/CMakeFiles/WBToolboxMex.dir/build.make:151: recipe for target 'lib/WBToolbox.mexa64' failed
make[2]: *** [lib/WBToolbox.mexa64] Error 1
CMakeFiles/Makefile2:254: recipe for target 'toolbox/CMakeFiles/WBToolboxMex.dir/all' failed
make[1]: *** [toolbox/CMakeFiles/WBToolboxMex.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
~~~

traversaro, 2018-06-04:
The problem is not occurring (and the WB-Toolbox works fine) if we comment out these lines: https://github.com/robotology/wb-toolbox/blob/devel/CMakeLists.txt#L20 .

diegoferigo, 2018-06-05:
@lrapetti Does everything work commenting out that line?

Can you try to restore that option and add [here](https://github.com/robotology/wb-toolbox/blob/08242faf30577e3e1bfd7a7996e1ad88d6092316/CMakeLists.txt#L20) the following line:

```cmake
target_link_libraries(WBToolboxMex pthread)
```

traversaro, 2018-06-05:
If this works, you can also use the [`FindThreads`](https://cmake.org/cmake/help/v3.5/module/FindThreads.html) module.

lrapetti, 2018-06-05:
> Can you try to restore that option and add here the following line:
> 
> target_link_libraries(WBToolboxMex pthread)

@diegoferigo 
I added that line here:
```
# Tweak linker flags in Linux
if(UNIX AND NOT APPLE)
   get_filename_component(LINKER_BIN ${CMAKE_LINKER} NAME)
   if(${LINKER_BIN} STREQUAL "ld")
       target_link_libraries(WBToolboxMex pthread)
       set(CMAKE_SHARED_LINKER_FLAGS "-Wl,--unresolved-symbols=report-all")
   endif()
endif()
# set(CMAKE_SHARED_LINKER_FLAGS "")
```

But I am getting the error:
```
lrapetti@iiticublap048:~/robotology-superbuild/build/robotology/WBToolbox$ make
CMake Error at CMakeLists.txt:20 (target_link_libraries):
  Cannot specify link libraries for target "WBToolboxMex" which is not built
  by this project.

-- Configuring incomplete, errors occurred!
See also "/home/lrapetti/robotology-superbuild/build/robotology/WBToolbox/CMakeFiles/CMakeOutput.log".
Makefile:278: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

```

traversaro, 2018-06-05:
`target_link_libraries(WBToolboxMex pthread)` need to be added **after** the WBToolboxMex target is created, you need to search where the command `add_library(WBToolboxMex ..)` or something similar is used and add the `target_link_libraries` **after** it.

diegoferigo, 2018-06-05:
@lrapetti Deeply sorry, I still had in the clipboard Sivio's link. I wanted to point you out [here](https://github.com/robotology/wb-toolbox/blob/08242faf30577e3e1bfd7a7996e1ad88d6092316/toolbox/CMakeLists.txt#L370).

lrapetti, 2018-06-05:
It didn't solve the original error, I'm still getting the same error message.

I have added `target_link_libraries` in `robotology-superbuild/robotology/WBToolbox/toolbox/CMakeLists.txt`:
```
## TOOLBOX S-FUNCTION MEX LIBRARY
## ==============================

if(WBTOOLBOX_USES_MATLAB)
    matlab_add_mex(
        NAME WBToolboxMex
        OUTPUT_NAME WBToolbox
        SRC include/base/SimulinkBlockInformation.h
        src/base/SimulinkBlockInformation.cpp
        src/base/Factory.cpp
        src/base/WBToolbox.cpp
        LINK_TO MxAnyType WBToolboxLibrary
        )

    target_link_libraries(WBToolboxMex pthread)
    target_compile_definitions(WBToolboxMex PUBLIC "MATLAB_MEX_FILE")
    target_include_directories(WBToolboxMex PUBLIC
        ${Matlab_INCLUDE_DIRS}
        ${Matlab_ROOT_DIR}/simulink/include)

    # Install S-Function
    install(TARGETS WBToolboxMex DESTINATION ${CMAKE_INSTALL_PREFIX}/mex)

    # Install TLC file
    install(
        FILES ${CMAKE_CURRENT_SOURCE_DIR}/WBToolbox.tlc
        DESTINATION ${CMAKE_INSTALL_PREFIX}/mex)
endif()

```

diegoferigo, 2018-06-12:
Removing that linker flag is a workaround of this issue, but it is not a solution. This is a Matlab problem, I open this issue open in order to track it, but we should investigate more in detail why Matlab complains about those missing symbols is his libraries.

I never heard of `libCppMicroServices.so`, and I didn't even know we were linking against it.

traversaro, 2018-06-12:
If indeed some Linux version of Matlab are shipped with this linking problem, I think removing the flags is (unfortunatly) the way to go. Matlab is well known to take ages to fix problems reported by users (they do not do maintenance release as far as I know).

S-Dafarra, 2018-10-30:
I got the same issue here. Just for future references the line to comment is this one: https://github.com/robotology/wb-toolbox/blob/da8951aa8c99bd50783a86c71b8d25ea316b4baf/CMakeLists.txt#L42

The link in the [comment above](https://github.com/robotology/wb-toolbox/issues/129?_pjax=%23js-repo-pjax-container#issuecomment-394371410) is outdated since the ``devel`` branch changed.

traversaro, 2018-10-30:
I think that `-Wl,--unresolved-symbols=report-all` is one of the flags that make sense to have in a development setup (similar to `-Werror`) but it would make sense to remove it from the default build, to avoid this not really controllable Matlab problems (similar to how `-Werror` should not be enabled by default, otherwise you risk that new compiler warnings break your compilation).

traversaro, 2018-11-14:
Friendly ping @diegoferigo .

traversaro, 2018-11-14:
Related issues: 
* https://stackoverflow.com/questions/50242393/linking-mismatched-stdthread-from-third-party-shared-objects
* https://it.mathworks.com/matlabcentral/answers/399797-linking-mismatched-std-thread-from-matlab-shared-objects

traversaro, 2018-11-14:
> This is a Matlab problem, I keep this issue open in order to track it, but we should investigate more in detail why Matlab complains about those missing symbols is his libraries.

I thought a bit about it, and I suspect this is not a Matlab problem. I imagine that Matlab 2018a ships a recent [`libstdc++`](https://gcc.gnu.org/onlinedocs/libstdc++/), but during the linking process we  use the system `libstdc++` instead of the `libstdc++` shipped with matlab. We link the system `libstdc++` for reasons similar to what we discuss in https://robotology.github.io/wb-toolbox/mkdocs/troubleshooting/: basically we want to link matlab libraries (linked with custom `libstdc++`) and our own superbuild-built libraries (built with the system `libstdc++`). `libCppMicroServices.so` is a library shipped with Matlab, and is expecting some symbols that I imagined are available in Matlab's `libstdc++` but are not available in the system one.

@lrapetti @S-Dafarra did you had this problems on Ubuntu 16.04 ? Hopefully this is not affecting the Ubuntu 18.04 `libstdc++`.

traversaro, 2018-11-14:
> @lrapetti @S-Dafarra did you had this problems on Ubuntu 16.04 ? Hopefully this is not affecting the Ubuntu 18.04's `libstdc++`. 

Indeed, the missing symbols are versioned (see https://www.akkadia.org/drepper/dsohowto.pdf for more info on symbol versioning) as `@GLIBCXX_3.4.22`. Checking [`libstdc++`'s manual on its ABI](https://gcc.gnu.org/onlinedocs/libstdc++/manual/abi.html) `@GLIBCXX_3.4.22`refers to symbols introduced in the `libstdc++` release corresponding to GCC 6.1 . Ubuntu 16.04 default compiled is GCC 5.3 , while Ubuntu 18.04 default compiled is GCC 7.3 (https://packages.ubuntu.com/xenial/gcc and https://packages.ubuntu.com/bionic/gcc)

lrapetti, 2018-11-15:
> @lrapetti @S-Dafarra did you had this problems on Ubuntu 16.04 ? Hopefully this is not affecting the Ubuntu 18.04's libstdc++

Yes, I was in Ubuntu 16.04

diegoferigo, 2018-11-17:
@traversaro Your explanation makes sense to me too. Here below is what I think we can do:

1. Disable the flag for missing symbols in Release, these kind of compilation failures, if harmless for the user as in this case, are not necessary for non-developers
1. Investigate if we still need to preload `libstdc++` on Ubuntu 18.04, it might be no longer necessary on updated matlab versions

diegoferigo, 2019-02-02:
The commit https://github.com/robotology/wb-toolbox/commit/31e585a9e8974537b545c8dc4512a243ed8160d1 enables the linker flag only in `Debug` builds. Closing.

closed 2019-02-02 by diegoferigo