robotology/wb-toolbox #139: Set References block gives an error in Position control mode

vibhoraggarwal, 2018-06-25:
**Description:**
- If I set position reference value as 0, then it stops to work and send me the error as mentioned below
- On any value except zero; although, it sets the position to the desired value, but gives the same error 

**Error shown**
```An error occurred while running the simulation and the simulation was terminated
Caused by:
Error reported by S-function 'WBToolbox' in 'test/Set References/S-Function':
Failed to set references.
```
cc : @diegoferigo

diegoferigo, 2018-06-25:
Which branch are you using?

vibhoraggarwal, 2018-06-25:
feature/motorparameter

diegoferigo, 2018-06-26:
Just to remove some variable here, does it happen also with the latest `devel`? You can switch now to `devel` since I merged the PR of that branch.

vibhoraggarwal, 2018-06-26:
Will check it with the devel and let you know.

vibhoraggarwal, 2018-06-26:
I switched to `devel` mode, and tried using the `set reference` block in `position mode`. The problem still exists.

vibhoraggarwal, 2018-07-17:
I got the green robot finally to work. I checked the yarp version on the head and the PC, which is `YARP version 3.0.100+20180629.11+git54d1442` and same on both.
The position reference still gives me the error.
`Failed to set references`

diegoferigo, 2018-08-03:
Some more details on this issue. I have the same problem on Gazebo with all the repos aligned with Yarp 3.1.

```
yarp: Port /WBT_synchronizer/rpc:o active at tcp://172.17.0.2:10052/
yarp: Sending output from /WBT_synchronizer/rpc:o to /clock/rpc using tcp

/home/dferigo/git/WB-Toolbox/toolbox/src/library/src/SetReferences.cpp@output:436
Failed to set references.
yarp: Removing output from /WBT_synchronizer/rpc:o to /clock/rpc
```

In short, line 371 fails. 

https://github.com/robotology/wb-toolbox/blob/e7998ffc391f876a991bcd20f9d0eaf4266827f9/toolbox/src/library/src/SetReferences.cpp#L352-L373

However, this should not be related to the toolbox itself but to the implementation of the interfaces. Strangely, `yarpmotorgui` works as expected.

cc @traversaro

traversaro, 2018-08-06:
> However, this should not be related to the toolbox itself but to the implementation of the interfaces. Strangely, `yarpmotorgui` works as expected.

I think that `yarpmotorgui` works because it does not check the return value of `positionMove` , see https://github.com/robotology/yarp/blob/f5e903f58cd4550504d6f6643e0bb3f91ece2ffe/src/yarpmotorgui/partitem.cpp#L1611 . Maybe the WB-Toolbox is the only frequently used client implementation that checks the return value, and this is why we are spotting the problem only here.

diegoferigo, 2018-08-06:
Probably @randaz81 can shed some light on this choice.

traversaro, 2018-08-06:
Yes, checking for the return value is the correct thing to do, I guess that there is no check in the `yarpmotorgui` just because nobody ever added it.

diegoferigo, 2018-08-06:
@randaz81 It used to work fine before YARP 3. I assume something in the implementation of the interfaces changed. On my setup things work in Gazebo, this leads me towards their implementation for the real iCub.

cc @valegagge @marcoaccame

traversaro, 2018-08-06:
Are we sure that the firmware on the robot is up-to-date?

diegoferigo, 2018-08-06:
The green iCub has currently the soft pid experimental branch since about two months. However, @vibhoraggarwal told me that this happens also on the purple (which status I'm not aware of). We're digging more on this issue, if any of you have some idea of a possible low-level culprit please let us know.

traversaro, 2018-08-06:
> Also Gazebo is affected. 

This seems to indicate that the problem is in the  `controlboardremapper` <---> `remote_controlboard` <---> `controlboardwrapper2` chain.

diegoferigo, 2018-08-06:
It is much simpler actually, i might have spotted the issue and it is something very very stupid...

What's wrong here? :sweat_smile: 

https://github.com/robotology/wb-toolbox/blob/c2211ae35807d2047b99068dbd25d159e82dee5a/toolbox/src/library/src/SetReferences.cpp#L345-L373

traversaro, 2018-08-06:
Missing `return` ?

traversaro, 2018-08-06:
No, `return` is there.. probably some tricky interaction of scopes, `switch` and `break` .

diegoferigo, 2018-08-06:
Fixed by #145

closed 2018-08-06 by diegoferigo