robotology/wb-toolbox #140: Yarp write block not working properly with the latest yarp devel

gabrielenava, 2018-07-05:
I moved to the latest `devel` branch in YARP and icub-main. I tried to use the WBToolbox block `yarp write`, but it seems it is not working correctly anymore. In particular, while I am still able to send values when using the command: `yarp write ... /portname`, I am not able to perform the same operation using the toolbox. Therefore even if I strongly suspect the issue is related to a recent modification done in `yarp`, I guess that the issue is limited to the WBToolbox block. 

Unfortunately, I didn't see any error message in Simulink or on the terminal related to this issue.

diegoferigo, 2018-07-05:
The problem is related YARP 3. The code of `YarpWrite` is very simple and right now it is not clear to me what could be the culprit.

diegoferigo, 2018-08-07:
Fixed by #146 @gabrielenava

closed 2018-08-07 by diegoferigo