robotology/wb-toolbox #143: Error using WBToolbox

vibhoraggarwal, 2018-08-03:
**Description:** 
I recently updated the WBToolbox in `devel` branch; but as I tried to run the simulations on Gazebo using WBToolbox in Simulink, I receive an error as mentioned below.

**Log:**
``` 
Error while obtaining sizes from MEX S-function 'WBToolbox' in 'TorqueModeV_2017b/synchronizer/GAZEBO_SYNCHRONIZER/Simulator Synchronizer'.
        Error using loadmodel (line 45)
        Unexpected Standard exception from MEX file.
        What() is:basic_string::_M_construct null not valid
```

@diegoferigo do you have any idea why it might be occuring?

diegoferigo, 2018-08-06:
@vibhoraggarwal No clue at first sight honestly, never had such issue so far. It looks a dangling reference of some sort, or a similar error.

This looks related to The Silvio Thing about locales (@traversaro) :sweat_smile: Can you reproduce it? We might look more in depth with the debugger.

vibhoraggarwal, 2018-08-06:
For a reference Matlab support team have described this error as well:
1. https://ww2.mathworks.cn/matlabcentral/answers/96898-why-do-i-receive-a-error-while-obtaining-sizes-from-mex-s-function-error-message
2. https://ww2.mathworks.cn/matlabcentral/answers/53964-strange-error-error-while-obtaining-sizes-from-mex-sfunction-in-simulink

diegoferigo, 2018-08-06:
We managed to catch the error which was very subtle.

In the big CMake refactoring 373924811968780993a14cc4b4444d6cfe272b91 the install path of the `WBToolboxLibrary.slx` changed from `WB-Toolbox` to `WBToolbox`. The documentation update is still pending and @vibhoraggarwal had his env variable pointing to the old folder. On a clean system this situation would trigger a matlab warning telling the user that the wb-toolbox's library couldn't be found. However, @vibhoraggarwal still had the old folder (`WB-Toolbox`) with an old copy of the simulink library, so this outdated version was read instead of the new one.

The switch to a plugin-based loading of the block library required changing all the block's S-Function parameters, including the name of the library containing its own class. The error displayed compiling the code in debug in the terminal from which Matlab was executed is the following:

```
/home/vaggarwal/robotology-superbuild/robotology/WBToolbox/toolbox/src/core/src/WBToolbox.cpp@mdlInitializeSizes:171
Factory error: loadlib/icub/left_leg/analog:o.so: cannot open shared object file: No such file or directory

/home/vaggarwal/robotology-superbuild/robotology/WBToolbox/toolbox/src/core/src/WBToolbox.cpp@mdlInitializeSizes:171
Factory error: loadlib/icub/left_foot/analog:o.so: cannot open shared object file: No such file or directory

/home/vaggarwal/robotology-superbuild/robotology/WBToolbox/toolbox/src/core/src/WBToolbox.cpp@mdlInitializeSizes:171
Factory error: loadlib/icub/right_leg/analog:o.so: cannot open shared object file: No such file or directory

/home/vaggarwal/robotology-superbuild/robotology/WBToolbox/toolbox/src/core/src/WBToolbox.cpp@mdlInitializeSizes:171
Factory error: loadlib/icub/right_foot/analog:o.so: cannot open shared object file: No such file or directory
```

The name of the library is populated by a random variable, in this case a port name. No clue why this behavior happens.

By cleaning the system and updating the matlab path we managed to get a working system.

closed 2018-08-06 by diegoferigo