robotology/wb-toolbox #149: Set Reference lock sets a fix default value for reference speed for joint movement in Position Control Mode

vibhoraggarwal, 2018-08-07:
### Description:
- **Not able to change the reference speed** for moving the joint in `Position Control Mode` 
- Instead the block sets a default value,`10 rad/sec`,  for reference speed.
- Changing the speed from `yarpmotorgui` is working though.

@diegoferigo

diegoferigo, 2018-08-08:
@vibhoraggarwal How did you check that the value didn't change? Note that the toolbox changes the reference speed only during the simulation, restoring the value found at its beginning when the simulation ends.

vibhoraggarwal, 2018-08-08:
> How did you check that the value..

@diegoferigo I observed the value in `yarpmotorgui` during the operation. The value was set to `10`

diegoferigo, 2018-08-10:
@vibhoraggarwal Can you please try the branch of PR #151? Those commits fix an unrelated error and on my setup the _Set References_ block works as expected. I do not exclude a strange link between the two given the randomness of the fixed bug.

diegoferigo, 2018-08-14:
@vibhoraggarwal Did you manage to try the fix?

vibhoraggarwal, 2018-08-15:
@diegoferigo Not yet. I plan to do it on Thursday.

vibhoraggarwal, 2018-08-16:
@diegoferigo I tried the branch of PR #151 on `icubGenova02`. But the problem still persists; the reference velocity is still set to a default value of `10`

diegoferigo, 2018-08-16:
I'll check on Monday on the robot. Similarly to #153 there is a mismatch between the simulated and the real robot.

vibhoraggarwal, 2018-08-20:
@diegoferigo I see that you merged the `diegoferifo/fix/plugin` branch to `devel`. Now I tried again with the updated devel; the position control mode successfully updates the reference velocity for the joint's motion now. 
Unless @diegoferigo want to mention something about the issue, we can close it.

diegoferigo, 2018-08-20:
Yes I merged the open PR since it was fixing a different (but related) bug. Probably something happened during checkouts when you did the tests last week because I didn't add any new commits. I'm glad to read that all works! :tada:

closed 2018-08-20 by diegoferigo