robotology/wb-toolbox #153: Problem with Set Reference block in Velocity Control Mode

FabioBergonti, 2018-08-14:
During some experiments I've tried to control the robot (iCubGenova02) in speed but it didn't move.
So I've done a very simple test, I've applied a sine as input for only one joint but it didn't work.
Using `yarp motorgui` I'm able to move joints in speed, so it seems that the following simulink block doesn't work
[image]

My WBtoolbox is currently update with [this commit](https://github.com/diegoferigo/WB-Toolbox/commit/44554cc743da733e74bebb24cc85f5bfea91efdb)

diegoferigo, 2018-08-14:
Does this happen also in Gazebo?
Do you have any kind of errors (either in Simulink or the terminal where Matlab was launched)?
Does the simulation stop or continue?

On that branch I tested only _Position_ and _Position Direct_.

FabioBergonti, 2018-08-14:
@diegoferigo @fiorisi

> Does this happen also in Gazebo?

No, in gazebo if I send a velocity signal the motors moves. But the simulator is not tuned, so the robot behaves in a strange way (also @GiulioRomualdi did some tests)
[image]

> Do you have any kind of errors (either in Simulink or the terminal where Matlab was launched)?

In matlab and simulink I didn't have errors. I didn't check the terminal.

> Does the simulation stop or continue?

It continued

> On that branch I tested only Position and Position Direct

A week ago I did also a test using *master* but it didn't work as well

diegoferigo, 2018-08-14:
Now `master` is quite old and works only with yarp 2. No one ever used velocity control as far as I know in the past and I am not surprised something was wrong.

If problems are only on the real robot with the new branch I fear we need to take some time for debugging them.

> For doing the tests on the real robot I've created a new branch where some .config have been modified in order to control iCub as R1

It would be nice having a Simulink model implementing a simple velocity control of a single joint. If changes of the configuration files are required, considering that the current status is not working, a PR targeting the gazebo model of robots-configuration is appreciated.

diegoferigo, 2018-08-20:
@FabioBergonti Are you sure you are using the right gains for the velocity control? I was told by @randaz81 that those values are very delicate and if they are wrong by order of magnitude the robot doesn't move. @GiulioRomualdi should have the right gains.

Another idea I had is that WBToolbox does not set the reference acceleration used under the hood to generate the velocity trajectory. However, the yarpmotorgui should set it to a very high value and running a Simulink model after opening the yarpmotorgui should assure that those values are set. I will fix this today, but I don't think it is the real cause.

diegoferigo, 2018-08-21:
If PR #156 does not fix the issue, the related bug fix will be included in the `4.1` release instead of the `4`.

FabioBergonti, 2018-08-21:
@diegoferigo 
> @GiulioRomualdi should have the right gains (are them maybe these ones you linked?).

Yes I think we have used the same values, because they are all 12 and 16 (as discussed in another issue with @hu-yue).

diegoferigo, 2018-08-21:
Let's talk f2f about those one.

Today @fiorisi and me did some experiment on the robot and a simple velocity controller worked as expected :tada: Closing.

closed 2018-08-21 by diegoferigo