robotology/wb-toolbox #161: Fail to compile QP block with Simulink

gabrielenava, 2018-09-20:
- [x] I already updated to the latest version I can use
- [x] I already checked similar issues using the search box
- [x] I already checked the website for known troubleshooting
- [x] I already cleaned my environment (by removing the wb-toolbox install folder and installing it again)
- [x] I already excluded the possibility that system configuration files (e.g. yarp related) might be responsible
- [x] I am sure the environment variables are correct
- [x] I tried to reproduce the error in simulation (if it affects the real robot)

#### Description

I tried to compile a [simulink model]() that contains the `QP` block. The compilation failed. Here is the only error available:

[image]

 
#### Platform

- `iCubGenova04` robot
- OS:  Ubuntu 18.04 
- Version:  `v4.0`
- Matlab Version: `2017b`

traversaro, 2018-09-20:
@gabrielenava In general, can you (also) copy-paste the error message instead of taking a screenshot? This make the error message searchable.

traversaro, 2018-09-20:
The error is in https://github.com/robotology/wb-toolbox/blob/ce58905b3aa5eb2b6f19da800f52fff9202290a6/toolbox/core/src/WBToolbox.cpp#L81 . Could you modify the code for printing also the result of the call to `platformSpecificLibName(data.first)` ?

traversaro, 2018-09-20:
And also `data.second` ?

gabrielenava, 2018-09-20:
I did the following modification:

```
         if (!factory || !factory->isValid()) {
	        std::cout <<  platformSpecificLibName(data.first) << std::endl;
		std::cout <<  platformSpecificLibName(data.second) << std::endl;
                wbtError << "Failed to create factory";
                return {};
            }
```

Now the Simulink error returns:

```
libWBToolboxLibrary.so
libQpOases.so

Error reported by S-function 'WBToolbox' in 'torqueBalancingYoga/controller_QP/Compute joint torques/QPSolver/One Foot/QP Two Feet':
Failed to create factory
Failed to get factory object

```

So I guess the to values you were looking for were the two `.so` files names?

traversaro, 2018-09-20:
The option `WBT_USES_QPOASES` was not enabled in the WBToolbox build.

closed 2018-10-03 by diegoferigo