robotology/wb-toolbox #167: [Announcement] Script to port controllers from WBT v4 to WBT v5

diegoferigo, 2019-01-21:
The upcoming `v5` release of WBT (at the time of writing in the `devel` branch) will break the compatibility of the controllers based on the previous releases.

I developed a [bash script](https://gist.github.com/diegoferigo/e2a693b9329c26e7f90d17cf1c3f014e) to automatically update the controllers. It should work [OOTB](https://www.wikiwand.com/en/Out_of_the_box_(feature)) in both GNU/Linux and macOS, and on Windows using the shell provided by [Git BASH](https://gitforwindows.org/).

It should be properly tested by @gabrielenava, please let me know if you face any issue.

cc @traversaro

lrapetti, 2019-03-12:
While trying to port a model to `WBT` using this bash script I am getting the following error:
```
$ ./WBT4_to_WBT5.sh -i torqueBalancingPHRIStandup.mdl -o torqueBalancingPHRIStandup_WBT5.mdl
\e[1;32m==>\e[0m Processing torqueBalancingPHRIStandup.mdl model
  \e[1;34m->\e[0m Creating copy of the input mdl
  \e[1;34m->\e[0m Processing wholeBodyStates blocks
sed: 1: "torqueBalancingPHRIStan ...": undefined label 'orqueBalancingPHRIStandup_WBT5.mdl'
```

The model I am trying to port is the one in https://github.com/Yeshasvitvs/whole-body-controllers/tree/andy_standup.

diegoferigo, 2019-03-12:
Are you using the bash shell?

diegoferigo, 2019-03-12:
I suspect the error is related to https://stackoverflow.com/questions/4247068/sed-command-with-i-option-failing-on-mac-but-works-on-linux. Can you please check?

lrapetti, 2019-03-12:
> I suspect the error is related to https://stackoverflow.com/questions/4247068/sed-command-with-i-option-failing-on-mac-but-works-on-linux. Can you please check?

I have tried to install and use `GNU-sed` (installed via `homebrew`) in place of MacOS `sed` and it ended successfully:
```
$ ./WBT4_to_WBT5.sh -i torqueBalancingPHRIStandup.mdl -o torqueBalancingPHRIStandup_WBT5.mdl
\e[1;32m==>\e[0m Processing torqueBalancingPHRIStandup.mdl model
  \e[1;34m->\e[0m Creating copy of the input mdl
  \e[1;34m->\e[0m Processing wholeBodyStates blocks
  \e[1;34m->\e[0m Processing wholeBodyModel blocks
  \e[1;34m->\e[0m Processing wholeBodyActuators blocks
\e[1;32m==>\e[0m The model compatible with WBT v5 has been created in torqueBalancingPHRIStandup_WBT5.mdl
\e[1;32m==>\e[0m Success
```

However, I still see those `Unresolved Link` blocks (while I can add new blocks without getting this error):
[image]

diegoferigo, 2019-03-12:
I know the problem. Unfortunately when you download the raw version of a file hosted in Gist, the url is unique and points to a specific revision. In fact, it was pointing to the first version of the script, I updated it to point to the most recent one.

Please follow again the instructions of the first post, it should work now.

lrapetti, 2019-03-12:
> I know the problem. Unfortunately when you download the raw version of a file hosted in Gist, the url is unique and points to a specific revision. In fact, it was pointing to the first version of the script, I updated it some while ago but `wget` was still downloading the outdated version.
> 
> Please follow again the instructions of the first post, it should work now.

@diegoferigo 
thanks! It seems that using the update conversion script it is working fine, and the new `.mdl` is loaded correctly.

gabrielenava, 2019-03-13:
I've ported the controllers inside [whole-body-controllers]() in the new WBT V5.0 and I tested the `momentum-based-yoga` controller on `iCubGenova04`. The demo was performing as before. I also tested all the controllers in simulation and I've found no issues related to the update.

Yeshasvitvs, 2019-03-21:
@diegoferigo correction in the example usage listed in https://github.com/robotology/wb-toolbox/issues/167#issue-401265345. The input model argument specified `-i` is missing.

`./WBT4_to_WBT5.sh -i model_WBT4.mdl -o model_WBT5.mdl`

closed 2019-04-25 by diegoferigo