robotology/wb-toolbox #174: Significant error in Gravity Torque

VenusPasandi, 2019-04-05:
I used the 'iCubGazeboV2_5 (no hands)' in the gazebo and control its position with 'set position'. Then, I compared the 'joint torques' and 'Gravity Torques' to check the gravity compensation. As you imagine, the Simulink model is simple as
[image]
The result is as
[image]
There is a significant error between the gravity force and the joint torques!

traversaro, 2019-04-05:
A few additional info:
* Which joints are you considering of the model (not the one that you are plotting, but all the joints that you specified in the WB-Toolbox configuration)? Remember that the joints you are not considering are assumed to be at their **zero** position w.r.t. to the computation of model quantities, including the gravity compensation torques. 
* Which model are you using in WB-Toolbox? Which model are you loading in Gazebo?

VenusPasandi, 2019-04-05:
> Let's move this issue in WB-Toolbox.

Thanks @traversaro , I didn't know where is the best place to open this issue!

> Which model are you using in WB-Toolbox? Which model are you loading in Gazebo?

I checked the models in gazebo and wbt, the both are the same.

> Remember that the joints you are not considering are assumed to be at their zero position w.r.t. to the computation of model quantities, including the gravity compensation torques.

Yes, This is the problem! Previously, I considered the joints that I wanted to control in `config` block. Now I consider all the joints in the countrol board and the joint torques and the gravity forces match as following
[image]

thanks @traversaro

closed 2019-04-05 by VenusPasandi

diegoferigo, 2019-04-17:
cc @anqingd