robotology/wb-toolbox #176: Compilation failure on Ubuntu 16.04

S-Dafarra, 2019-05-29:
- [ ] I already updated to the latest version I can use
- [ ] I already checked similar issues using the search box
- [ ] I already checked the website for known troubleshooting
- [ ] I already cleaned my environment (by removing the wb-toolbox install folder and installing it again)
- [ ] I already excluded the possibility that system configuration files (e.g. yarp related) might be responsible
- [ ] I am sure the environment variables are correct
- [ ] I tried to reproduce the error in simulation (if it affects the real robot)

#### Description

When trying to compile WB-Toolbox from a plain installation inside ``robotology-superbuild``, it fails with the error down below
#### Platform

Ubuntu 16.04

#### Reproducibility

Describe the steps for reproducing the behavior. If possible, upload a simple Simulink model (only if the error does not either involve the real robot or it is reproducible in simulation).

1. First step
2. Second step
3. ...

#### Screenshots

If applicable, add screenshots to help explain your problem (e.g. Simulink Model, Scopes, etc.)

#### Additional information

- OS: Ubuntu 16.04
- Version / Commit: https://github.com/robotology/wb-toolbox/commit/bbd3e52ce53a705498537bc99410c59619c46fb2

#### Additional context

```
[ 44%] Performing build step for 'WBToolbox'
[  3%] Generating code from clock_rpc.thrift
Error copying file "/home/sdafarra/Software/robotology-superbuild/build2/robotology/WBToolbox/deps/thrift/CMakeFiles/yarpidl_thrift/ClockServer.h" to "/home/sdafarra/Software/robotology-superbuild/build2/robotology/WBToolbox/deps/thrift/include//ClockServer.h".
deps/thrift/CMakeFiles/ClockRpc.dir/build.make:62: recipe for target 'deps/thrift/src/ClockServer.cpp' failed
make[5]: *** [deps/thrift/src/ClockServer.cpp] Error 1
make[5]: *** Deleting file 'deps/thrift/src/ClockServer.cpp'
CMakeFiles/Makefile2:135: recipe for target 'deps/thrift/CMakeFiles/ClockRpc.dir/all' failed
make[4]: *** [deps/thrift/CMakeFiles/ClockRpc.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/WBToolbox.dir/build.make:106: recipe for target 'robotology/WBToolbox/CMakeFiles/YCMStamp/WBToolbox-build' failed
make[2]: *** [robotology/WBToolbox/CMakeFiles/YCMStamp/WBToolbox-build] Error 2
CMakeFiles/Makefile2:961: recipe for target 'CMakeFiles/WBToolbox.dir/all' failed
make[1]: *** [CMakeFiles/WBToolbox.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

S-Dafarra, 2019-05-29:
Sorry, when creating another folder I did not realize it was linking the wrong ``yarp`` and ``GazeboYarpPlugin`` version. I cleaned up the setup and it compiled.

closed 2019-05-29 by S-Dafarra

traversaro, 2019-05-29:
Just to understand, why you need to install on Ubuntu 16.04 ? We are progressively dropping support for it along the stack (see http://wiki.icub.org/wiki/YARP_Supported_Distributions) so if for any reason we need to support Ubuntu 16.04 in special cases it would be important to know which part of the software we need to support.

S-Dafarra, 2019-05-29:
I would not say it is a special case. It seemed not to be an issue related to OS tough.

traversaro, 2019-05-29:
>  It seemed not to be an issue related to OS though.

Yes, the question was actually independent from the issue.