robotology/wb-toolbox #179: Simulink time and yarp time diverge

VenusPasandi, 2019-07-03:
I am using Simulink with Gazebo. I noticed that the Simulink time and Yarp time are not coherent. In fact, Simulink always has a delay with respect to Yarp time. I checked the synchronization with a simple Simulink which I am sure does not need much time for running as the following,
[image]
So, I don't get or send any data to the robot and just read Yarp time. The comparison of Yarp Time and Simulink time is interesting,
[image]
As @FabioBergonti suggested I tried this simulation also on the @Giulero 's laptop and the result was the same. 
I also tested this simulation with the real robot (i.e. without gazebo). In this case, the Simulink and the Yarp time are coherent as the following
[image]
I can not understand what is the problem! Is it any problem with Gazebo? Is it so heavy that consumes such amount of memory that even a simple Simulink file takes much time to run?

traversaro, 2019-07-03:
I think this is normal due to how the `SimulatorSynchronizer` works: as you can see in https://github.com/robotology/wb-toolbox/blob/master/toolbox/library/src/SimulatorSynchronizer.cpp#L204 , the logic is the  following: 
* If I understood correctly, in this case the `Yarp Time` is the System monotonic Clock.
* First, the Simulink model runs, until it is time to run the `SimulatorSynchronizer` block. During this time, the Gazebo simulation is **paused**
* Then, the   `SimulatorSynchronizer` block's `output` runs, and the `stepSimulationAndWait` blocking function is called. During this time, Gazebo runs and the method returns when Gazebo have been simulating for enough time. During this time, Gazebo has its own real-time synchronizer that is running, so it pauses to make sure that the simulated time passed is similar to the real time passedand the Simulink model is not running because it is blocked by the  `stepSimulationAndWait` call.
* Due to the network delay and scheduler delay intrinsic in starting/stopping the Gazebo simulator, and due to the fact that the Gazebo simulator and the Simulink integration never run simultaneously, the "Simulink" time always lag a bit behind the system clock.

traversaro, 2019-07-03:
Note that this kind of problems would be  avoided by using some more proper form of Co-Simulation (https://dl.acm.org/citation.cfm?id=3179993) between the Simulink simulation and the Gazebo simulation, such as the one described in https://github.com/robotology/wb-toolbox/issues/164 .

VenusPasandi, 2019-07-03:
thank you @traversaro for the detailed explanation. Accordingly, when we use Simulink and Gazebo, the incoherency of `Yarp Time` and `Simulink Time` does not mean that Simulink and Gazebo are not synchronized. Though, in the case of `Real Time Synchronizer`, if `Yarp Time` and `Simulink Time` are not coherent, it means that Simulink is not synchronized with the robot motion. Am I right?

diegoferigo, 2019-07-03:
> Though, in the case of Real Time Synchronizer, if Yarp Time and Simulink Time are not coherent, it means that Simulink is not synchronized with the robot motion. Am I right?

Yes, the real time synchronizer tries to enforce the matching between the simulink and real time. The Yarp time is synchronized only if you don't use any network clock, situation quite uncommon that can be ignored.

In the past we had few problems of small mismatches between simulink time and real time #160 (solved bypassing simulink gui), but in general synchronization is very good as you have shown.

VenusPasandi, 2019-07-04:
Thank you @diegoferigo, I got my answer.

closed 2019-07-04 by VenusPasandi